victronenergy/venus #765: venus-large: varia

mpvader, 2021-01-06:
I'll use this issue to keep note of various things I come accross while working with the large images.

Shouldn't this be changed the venus-swu-large?

```
root@einstein:~# cat /etc/venus/swu-name 
venus-swu
```

mpvader, 2021-01-07:
Added (2).

nmbath, 2021-01-11:
Good evening. 

I am not sure if the attached logs will help with point 2. I have managed to use software update on a Cerbo without problem. My Raspberry Pi fails, even when I rename the file to remove the -large.

The filenames that failed on the Raspberry Pi are:
venus-swu-raspberrypi2-20210111135948-v2.70~3-large-10.swu
venus-swu-raspberrypi2-20210111135948-v2.70~3-10.swu

Even when I manually request the upgrade using these commands it fails

/opt/victronenergy/swupdate-scripts/check-updates.sh -swu /media/sda1/venus-swu-raspberrypi2-20210111135948-v2.70~3-large-10.swu

or 

swupdate -v -i venus-swu-raspberrypi2-20210111135948-v2.70~3-10.swu -e "stable,copy2"

It seems to be something in the embedded file venus-image-large-raspberrypi2.ext4.gz  that is causing an issue.

[Cerbo - Success.txt](https://github.com/victronenergy/venus/files/5798381/Cerbo.-.Success.txt)
[Raspberry Pi - Failed.txt](https://github.com/victronenergy/venus/files/5798391/Raspberry.Pi.-.Failed.txt)

mpvader, 2021-01-11:
Oh yes. I’m pretty sure you’ll need to reflash the rpi - unfortunate as that may be. We changed some things making the old flashed cards incompatible. See here:

https://github.com/victronenergy/meta-victronenergy/commit/892d2c117e46239baa095f85b2c2f94b3d36b75f

mpvader, 2021-01-11:
Ps; needs mentioning in changelogs ofcourse. Thanks for bringing it up! I’ll make sure and then add it

mpvader, 2021-06-28:
Fixed

closed 2021-06-28 by mpvader