victronenergy/venus #768: Changing System Name should also change hostname (/etc/hostname)

Knobee, 2021-01-11:
When the system name is changed:

[Settings -> System Setup -> System Name]

This should also change the hostname on the Raspbian side (put the System Name into /etc/hostname).

Without this change, the DHCP server will continue to use the "raspberrypi2" name as the image is shipped.  Some DNS servers auto-populated from the DHCP server and it would be nice to be able to connect to "victron.boat.example.com" instead of "raspberrypi2.boat.example.com"

This change would also necessitate cleaning the system name of all non-alphanumeric characters (no spaces, slashes, etc)

kwindrem, 2021-01-11:
I agree it would be nice to change the hostname. Matching the System Name is as good a way as any except that spaces and other "special" characters need to be addressed. I'm not sure restricting the System Name to avoid spaces and other characters that might mess up the hostname is the best idea. Spaces could be replaced with underscores, etc.

Note that there is code somewhere within Venus that changes /etc/hostname on each boot. I made a manual change and Venus put it back.

jhofstee, 2022-05-21:
If you want to do lookup by names I guess it would be better to setup your router to assign a fixed ip and setup your DNS resolver accordingly instead of trusting names in a DHCP requests. If you want to have this behavior it can be done by changing /etc/hostname. I don't see a need for Venus to do it in general.

closed 2022-05-21 by jhofstee