victronenergy/venus #769: Node Red and Relay Control

nmbath, 2021-01-11:
I have just got my Cerbo onto the latest large image (v2.70~3-large-10).

I have been looking at the relay control, and really like the fact that the MPPT and BMV Relays are available. I have noticed the following however.

1. SmartShunts are shown in the list even though they do not have a relay. The device "12v House Loads" is a SmartShunt. Also no matter how I configured the relay on the BMV (Default, Charge Mode, Remote) I could change the relay. Not sure if this is by design or sensible of the relay in Default or Charge Mode has configurations set. 
2. Setting the relay in an MPPT (Connected via VE.Direct) did not work. It didn't matter what mode I set the relay in on the MPPT, I just couldn't get it to trigger.

[image]

Great capability and look forward to making use of them.

mpvader, 2021-01-11:
Hi.

1. hiding devices that dont have a relay from the list would indeed be nice.
2. I’d expect the BMV relay to work with large image 10. It includes an updated driver for the vedirect port, which adds below two changes. Could you try with dbus-spy? And perhaps otherwise see where the issue is?
    - bmv: Add /Settings/RelayMode (=VE_REG_RELAY_MODE).
    - bmv: Allow setting /Relay/0/State when /Settings/RelayMode is REMOTE(=2)
3. MPPT Relay still doesn’t work; details in the Venus OS large manual, section “controlling relays”

nmbath, 2021-01-11:
@mpvader, Thank you or the response. Something must have been messed up with my system during upgrades. I have rebooted my Cerbo and can confirm that the BMV values are there and I can only change the relay now when the /Settings/RelayMode is 2.

Playing around further it would be useful if we can have an input for the relay state. This will allow for easy creating of flows that can for instance turn a relay on when the temperature drops below x, and off when it has risen above y.

I look forward to the MPPT software update for the manual relay mode.

Interesting that someone has reported relays working on the Phoenix Smart Charger as I did not see that in the manual list when I just looked.

Once again many thanks for this feature set as it will be very valuable and offer a load of flexibility.

nmbath, 2021-01-13:
OK, while looking at other things, I noticed in the node-red-contrib-victron repository a [change in May 20](https://github.com/victronenergy/node-red-contrib-victron/commit/61db660cf7a669c9cd24308fbcccff258adcc365) to add what looks like input relay state.

However do not see and nodes listed for relay inputs. Have I missed something?

mpvader, 2021-01-13:
Yes, the version that added that (v1.4.0) was never included in a build yet. But now it will be.

Hey, I'll close this issue here; since I want to not have too many node-red issues in our main repo.

Please test the upcoming build; and if there are still open things regarding node-red-contrib-victron, then please open an issue here: https://github.com/victronenergy/node-red-contrib-victron/issues

closed 2021-01-13 by mpvader