victronenergy/venus #776: v2.62 possible wifi issue

mpvader, 2021-01-28:
Details in the email

mansr, 2021-01-28:
Possibly relevant messages:
```
Jan 27 15:55:24 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Trying to associate with 24:a7:dc:1a:51:02 (SSID='EWKY' freq=2412 MHz)
Jan 27 15:55:24 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Association request to the driver failed
Jan 27 15:55:28 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Trying to associate with 20:47:ed:dd:99:96 (SSID='EWKY' freq=2412 MHz)
Jan 27 15:55:28 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Association request to the driver failed
Jan 27 15:55:32 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Trying to associate with 38:a6:ce:01:30:6e (SSID='EWKY' freq=2462 MHz)
Jan 27 15:55:32 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Association request to the driver failed
Jan 27 15:55:32 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Trying to associate with 24:a7:dc:1a:51:02 (SSID='EWKY' freq=2412 MHz)
Jan 27 15:55:32 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Association request to the driver failed
Jan 27 15:55:36 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Trying to associate with 20:47:ed:dd:99:96 (SSID='EWKY' freq=2412 MHz)
Jan 27 15:55:36 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Association request to the driver failed
Jan 27 15:55:40 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Trying to associate with 38:a6:ce:01:30:6e (SSID='EWKY' freq=2462 MHz)
Jan 27 15:55:41 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Association request to the driver failed
Jan 27 15:55:45 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Trying to associate with 24:a7:dc:1a:51:02 (SSID='EWKY' freq=2412 MHz)
Jan 27 15:55:45 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Association request to the driver failed
Jan 27 15:55:50 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Trying to associate with 20:47:ed:dd:99:96 (SSID='EWKY' freq=2412 MHz)
Jan 27 15:55:50 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Association request to the driver failed
Jan 27 15:55:55 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Trying to associate with 38:a6:ce:01:30:6e (SSID='EWKY' freq=2462 MHz)
Jan 27 15:55:55 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Association request to the driver failed
Jan 27 15:56:00 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Trying to associate with 24:a7:dc:1a:51:02 (SSID='EWKY' freq=2412 MHz)
Jan 27 15:56:00 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Association request to the driver failed
Jan 27 15:56:00 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="EWKY" auth_failures=1 duration=10 reason=CONN_FAILED
Jan 27 15:56:10 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: CTRL-EVENT-SSID-REENABLED id=0 ssid="EWKY"
Jan 27 15:56:10 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Trying to associate with 20:47:ed:dd:99:96 (SSID='EWKY' freq=2412 MHz)
Jan 27 15:56:10 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Association request to the driver failed
Jan 27 15:56:10 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="EWKY" auth_failures=2 duration=20 reason=CONN_FAILED
Jan 27 15:56:30 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: CTRL-EVENT-SSID-REENABLED id=0 ssid="EWKY"
Jan 27 15:56:30 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Trying to associate with 38:a6:ce:01:30:6e (SSID='EWKY' freq=2462 MHz)
Jan 27 15:56:30 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Association request to the driver failed
Jan 27 15:56:30 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="EWKY" auth_failures=3 duration=30 reason=CONN_FAILED
Jan 27 15:57:04 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: CTRL-EVENT-SSID-REENABLED id=0 ssid="EWKY"
Jan 27 15:57:04 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Trying to associate with 24:a7:dc:95:5b:1a (SSID='EWKY' freq=2462 MHz)
Jan 27 15:57:04 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Association request to the driver failed
Jan 27 15:57:04 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="EWKY" auth_failures=4 duration=60 reason=CONN_FAILED
Jan 27 15:58:36 beaglebone daemon.warn connmand[1204]: Skipping disconnect of 45574b59_managed_psk, network is connecting.
Jan 27 15:58:40 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Trying to associate with 24:a7:dc:1a:51:02 (SSID='EWKY' freq=2412 MHz)
Jan 27 15:58:40 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Association request to the driver failed
Jan 27 15:58:40 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="EWKY" auth_failures=1 duration=10 reason=CONN_FAILED
Jan 27 15:58:54 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: CTRL-EVENT-SSID-REENABLED id=0 ssid="EWKY"
Jan 27 15:58:54 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Trying to associate with 20:47:ed:dd:99:96 (SSID='EWKY' freq=2412 MHz)
Jan 27 15:58:54 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Association request to the driver failed
Jan 27 15:58:54 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="EWKY" auth_failures=2 duration=20 reason=CONN_FAILED
Jan 27 15:59:18 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: CTRL-EVENT-SSID-REENABLED id=0 ssid="EWKY"
[... cut ...]
Jan 27 16:33:30 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: CTRL-EVENT-SSID-REENABLED id=0 ssid="EWKY"
Jan 27 16:33:30 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Trying to associate with 20:47:ed:dd:99:96 (SSID='EWKY' freq=2412 MHz)
Jan 27 16:33:30 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: Association request to the driver failed
Jan 27 16:33:30 beaglebone daemon.notice wpa_supplicant[1259]: wifi0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="EWKY" auth_failures=4 duration=60 reason=CONN_FAILED
```

mpvader, 2021-01-30:
Yesterday, I installed v2.62~14 on the system showing above; now waiting for Mark to confirm that its fixed.

mpvader, 2021-02-07:
Finally this came down to this: after selecting "forget network" in the menu, it's impossible to connect again without rebooting first.

Reproducible on a Venus GX on v2.62~beta something, v2.60, v2.58 as well, using an EasySolar Wifi as the access point.

Restarting connman and wpa_supplicant makes no difference

Unloading and reloading the rtl8723bu module fixes it

Above doesn't happen with the new wifi module (6223E-UUC).

conclusion: no regression, and also not something we'll fix, as its either in the hardware of the wifi driver.

closed 2021-02-07 by mpvader