victronenergy/venus #778: Method of Tracking Relay Functions

nmbath, 2021-02-01:
Victron,

Would it be possible to have a way of requesting and tracking relay functions. As the system is Open, people can create new uses for the relay and need to define new Functions. As both Victron and others define these uses, code can break if there is no single defined means of tracking function assignments.

Having a way tor request a relay function and using that value will mean that two functions in the future cant clash and have problems.

Similar request as [747](https://github.com/victronenergy/venus/issues/747) for Digital Input functions.

Could be as simple as maintaining a record in a file within git.

izak, 2021-02-02:
@nmbath , I would imagine that using the relay for functions other than already catered for (generator start/stop, pump start/stop, alarm) would require either control through modbus/mqtt, or your own custom software that accesses the relay through `com.victronenergy.system/Relay/0/State`, and as such would simply require the relay function to be set to manual? It seems pointless to me to add more functions if they essentially equate to "user defined". Do I understand this issue correctly?

nmbath, 2021-02-02:
@izak, 

I think you do understand the issue. One needs to ensure if a 3rd party creates a driver to control the relay through com.victronenergy.system/Relay/0/State, a unique function id is required.

I did indeed write a driver to control a relay based on temperature and controlling it through  com.victronenergy.system/Relay/0/State or com.victronenergy.system/Relay/1/State. I am aware that NodeRed might remove the need to create such drivers in the future, but none of us have a crystal ball and able to state that a dedicated driver for a given capability would not be better than NodeRed. Setting the relay function to manual, does not allow automated decision based switching off the relay via a driver.

I noticed that Victron has created some new capability that has used ID's 4 and 5 for CD and LD. I had used these exact values in my drivers. https://github.com/victronenergy/venus/issues/737. While Victron decided to do this within a driver rather than NodeRed, just goes to show how a systems integrator or component developer might feel a driver is the better solution.

I will off course have to go and change my code, which is not a problem. But of course as this is open source anyone can create functions and drivers. Therefore it seems sensible to have a way of maintaining a defined list or maybe even some reserved values that people can use and not then have to change code at a later date or track Victron change logs to know a change would be needed to their code.

Clearly if Victron does not see any value of having a means of tracking and ensuring drivers have a unique value and in affect keep this part closed I understand.

jhofstee, 2022-05-21:
I would guess that simply setting the relay to manual mode would be sufficient for all / most cases. Since it is no longer an issue for @nmbath I will close the issue.

closed 2022-05-21 by jhofstee