victronenergy/venus #779: Add Wakespeed alternater regulator monitoring

mpvader, 2021-02-04:
### Steps:

- [ ] Add it to vecan-dbus driver, after which it will be available on D-Bus
- [ ] Add it to the gui, so its visible in the gui
- [ ] Add to VRM Portal and vrmlogger
- [ ] Add to HTML5 Application, used on Marine MFDs
- [ ] Add to ModbusTCP
- [ ] Anything else?

### Specs

Protocol will be NMEA2000.

Required information to display:

| Field | Origin | Notes |
| ----- | ------ | -----| 
| RPM                    | Field 1 in PGN 127488: Engine Parameters, Rapid Update     |  |
| Alt. temperature |  PGN 127508 battery status, under the Alternator instance   |   |
| Load %   |  @thomasonw where is this?                                                                                 |   |
| Charge state (bulk, float etc)   |  PGN 127507 Charger status      |   |
|   |  @thomasonw anything else?   |   |   

These fields are available, but we won't include them:

- Battery state of charge: isn't calculated by the Wakespeed
- Battery current: only available if the shunt is wired in series with the alternator, which is normally isn't.
- Battery voltage: that one we could add - @Al?
- Battery configuration (chemistry, capacity, etc): not relevant, configuration is out of scope
- Charger configuration: also not relevant: not relevant, configuration is out of scope

Why not RV-C since the Wakespeed supports RV-C as well? Thats because NMEA2000 and RV-C are mutually exclusive, and rest of Victron system is also on NMEA2000.

Documentation, including information about the used instances and details of all available PGNs:

- [Wakespeed Communications and Configuration Guide v2.3.0.pdf](https://github.com/victronenergy/venus/files/5925653/Wakespeed.Communications.and.Configuration.Guide.v2.3.0.pdf)
- [WS500 Quick Start.pdf](https://github.com/victronenergy/venus/files/5925654/WS500.Quick.Start.pdf)

mpvader, 2021-02-04:
added battery current to above

thomasonw, 2021-02-04:
@mpvader  @jhofstee 1st of two replies:

PGNs to utilize:  Approach is to use the N2K type messages (as opposed to native RV-C messages) from the WS500.  Users will need to note this and assure they do not turn off the N2K messages in the WS500 configuration.  

Instance number details:  WS500 uses two 'Instance numbers':  Battery and charger.

- Battery:  Which battery is the WS500 / alternator attached to?  Battery Instance is used to send out the WS500's understanding of battery condition:  Vbat, Abat, SOC, temperature, etc.  These values are either measured using the WS500 sensors, or reflect what has been transmitted to it via CAN from a remote BMS.  In the case where BMS Aggregation is active (the WS500 is listing to more than one BMS in parallel), the 'Battery' messages will send out the WS500's aggregated view of the entire battery system, not each individual BMS's data.

- Charger:  What 'Charger' instance is this?  WS500 uses 'Charger' instance when sending out condition of the WS500/Alternator.  RPMs, Temperature, etc.   Some fields may be populated with the 0xFF indicator if they are not available.  One key example:  Current.  If the WS500 is configured with the 'Shunt At Bat' flag set (default, and most common deployment), then the Current Field in these 'Charger' instance messages will be flagged with 0xFF/0xFFFF indicating the data is not available.

Battery Instance is the Battery ID the alternator is associated with (e.g., the DC Bus the alternator is connected to) and defaults to 0 (House Battery). 

Charger Instance reflects the WS500 'Charger Instance' ORed with 0x30 to indicate an 'Alternator'** .  Note: In a multi-alternator install each WS500 will have its own 'Charger Instance' either set by the installer at time of configuration, or auto-incremented if the WS500 detects another WS500 present and there is an instance number conflict.  And in all cases if both  WS500's are attached to the same battery, they will be configured with the same Battery Instance.  **The ORing of 0x30 to a Device Type is how the Charger Instance is used in RV-C, and has been carried forward into the N2K messages for consistency.   However, perhaps this is not the best approach?  I welcome feedback on this detail, as many N2K messages already carry a 'Charger Device Type' to indicate Alternator  (Vs. say a DCDC charger, or Solar..)

Instance numbers may be changed at time of install if the default values are not appropriate.  

The WS500 primarily operates using RV-C CAN protocol, where instance numbers are 1 origin (0 == fault), while N2K is 0 origin based.  As such when configuring the WS500, the default Battery and Charger instances are both = 1.  Then when the WS500 transmits N2K messages, is sends all instances as (n-1), so from the N2K stand point default instances for the battery and charger will be  = 0.

There seems to be a convention in N2K to reuse messages for different purposed.  As an example:  The N2K message 'DC Status'  (PGN127508) is sent twice:

- PGN127508  with Instance = 0 (Default battery instance indicating house battery):  Transmits the WS500 understand of the battery condition:  Voltage, Current, Temperature.

- PGN127508  with Instance = 0x30 (Default Charger instance + 0x30 to indicate alternator):  Transmits the WS500 measured Voltage, Current and Temperature.  (If known, see comment about Alternator Current above).   If there was a 2nd alternator present, it would also transmit a 127508 message, with the Instance = 0x31

Independent SIDs are used for Battery orientated messages vs. Charger orientated messages.

Here is an edited version of the PGN table with notes:

Field | Origin
-- | --
Device Details| PGN 126996 Product Information,  TJC Micro Manufactures ID (SAE assigned =  963 ), WS500 Product Code (WS500 = 200, WS3000r = 350), firmware version, Device Number (Internal serial number).
Device Name | PGN 126998 Configuration Information,  Device Name as configured by user (ala, Port Alternator, Sbd Alternator) --> (Field #1 IIRC).  Wakespeed URL /  manufacturer information.  (Field #3)
RPM | Field 1 in PGN 127488: Engine Parameters, Rapid Update  (Derived from Stator sensing or engine J1939 data if WS500 is connected to engine CAN bus)
Alt. Voltage| PGN 127508 DC status, under the Alternator instance (Very rare for this to be different then Battery Voltage, only possible if the BMS is able support Remote Instrumentation of Battery Voltage).  N2K Device Type field is set as Alternator
Alt. Amperage | PGN 127508 DC status, under the Alternator instance.  If regulator is configured with the Shunt Sensing wires on an alternator shunt, as opposed to a battery shunt, and the flag 'Shunt At Battery' is set false, this field will be populated with measured Alternator Current.  Else it will be populated with 0xFFFF.  We do have some installs done in this way, and it depends on Remote Sensing of the battery current either via the BMS or perhaps a dedicated battery sensor to back-fill the battery current.   N2K Device Type field is set as Alternator
Alt. temperature | PGN 127508 DC status, under the Alternator instance  (A detail, if the ATS is damaged or not installed, this field will be set to 0xFF) N2K Device Type field is set as Alternator
Alt. temperature | PGN 127750 Charger status.  (Perhaps better source then 127508??)  Uses Alternator Instance, same comment about missing ATS as 12508
[... cut ...]
I do believe that is about it for this 1st message.  There are some overlaps between different N2K messages, but information is the same.   And a side note, v2.4.0 firmware (to be released very soon) has a new option where the WS500 is 'Powered On' all the time, sending out these CAN messages, and the 'Key-switch' is then routed in via the Feature-in port.  Not sure how common this will be used in pure Alternator Only installs of the WS500, but it gives the option of the WS500 transmitting battery details even when the engine has been turned off.

thomasonw, 2021-02-04:
2nd response, to some of the other questions.

_" Battery voltage: that one we could add - @thomasonw any particulars? or always available?"_

The WS500 will transmit as 'Battery Voltage' what it senses on the VBat+ and VBat- wires.  There is a capability that is very uncommon (only one BMS I know if is able to support this) where those sensing wires may be attached locally to the alternator and the BMS remotely transmits battery voltage. 

_"Wakespeed only powered up when ignition is present? @thomasonw pls confirm."_

In installations to date, this is true.  The Brown 'Ignition' wire must be powered to allow the WS500 to operate.  However, with the (later this week?) release of firmware 2.4.0 a new option will be enabled where the Brown wire may be powered 100% of the time keeping the WS500 active at all times - and in this case the 'Ignition' is then connected to the Feature-in port to signal when the alternator should be driven.

_"Why not RV-C since the Wakespeed supports RV-C as well? Thats because NMEA2000 and RV-C are mutually exclusive, and rest of Victron system is also on NMEA2000."_

RV-C and NMEA2000 are both based on J1939, however they are not purposely designed to be compatible.  I know of one (maybe more) PGN overlaps where RV-C and NMEA2000 use a given PGN with different meanings.   In my experience to date, this has caused no issues in the field - but the conflict is there.  And given that we can accomplish this 'integration' using N2K like messages (as noted in the prior posting), this seems a good way to go.  As a side note, though I am not aware of any conflicts up to date, if there is an install where the RV-C messages causes an issue with the N2K devices, the option of using a dedicated CAN port is always available, as well as using a CAN-to-CAN bridge with filtering (One example is detailed in our WS500 Commutations and Configuration guide, noted above).   --OR-- one could configure the WS500 to disable RV-C, but that should be done with caution, as it will prevent regulators talking to each other in a twin engine install as well as disable the ability to interact with BMS's via CAN.

nmbath, 2021-02-04:
To add to what @thomasonw has shared. 

The common installation is to have the shunt at the battery, even though it can be configured at the alternator. This allows the WS, or any other alternator controller for that matter, to supply loads when the battery is charged. It does it by regulating the current through the battery shunt and keeping it to zero. Further this allows it to supply loads when for example battery charging should not be aloud through ATC and low temperature. 

In my setup I have seen the Alternator react to significant loads from the inverter and not seen the voltage sag below float or incremental current into the batteries. 

As most installs I have done, seen and read about configure the shunt at the battery, I concur that it is common.

A second consideration regarding shunt location, is that a lot of engines use the engine block as ground and thus measuring current from the alternator would require the shunt to be in the positive side of the circuit, or have the engine starting load flow through the shunt as it is installed between the engine block and the negative battery bus. That then creates more problems with multiple alternators and systems that have a stock alternator at 12v and additional alternator at 24/48v.

I am sure we can solve the problem, but requires a lot more thought.

Consideration needs too be given to dual alternator and controller installs. The Wakespeed is able to be run in a master/slave configuration. 

You can look at the WS and it can run in the following modes

- Standalone controlling an alternator
- Standalone controlling a DC-DC converter
- Master / Slave controlling separate alternators
- Master / Slave with one controlling an Alternator and one controlling a DC-DC converter

nmbath, 2021-02-04:
@mpvader, 

I have been doing some prep work to create a driver and extract performance data from the WS and insert into dbus and then present in some QML screens. I ran into a challenge and posted in Victron-dev-venus to see if there is a way around it.

One should consider both Ve.Can and BMS.Can.

Be more than happy to collaborate and help.

VETWorks, 2021-08-17:
Any updates to the Wakespeed WS500 integration? It would be great to have this natively supported. Maybe add master/slave in a later update, but get a working version out that would be applicable on most installs?

Rick-OGSS, 2021-10-09:
Hey mpvader,  we are a Wakespeed dealer and are watching this project closely.  We have a full test lab set up with all the Victron and Wakespeed equipment needed to perform some alpha/beta testing for you if you like.  Our customers are biting at the bit for this integration, so anything we can do to help, just let us know.
Rick OGSS

Tuckie, 2022-02-23:
Any updates on this? Very interested in a WS3000 integration.

martinbosma, 2022-05-06:
Included in vecan-dbus v2.89 and gui changes are ready. 
@mpvader How about the other tasks from the main post (vrm, vrm-logger, html5, modbus, etc)?

mpvader, 2022-05-06:
Lets start with having it in VRM (advanced page only) and modbus.

@izak

millijuna, 2022-05-07:
Hi All, would be happy to help test this. We’re just building out a system with a WS500, Cerbo GX, and REC ABMS. I have the REC and WS500 connected to the BMS.CAN interface (running at 500kbps), rather than to the VE.Can (and the rest of my NMEA-2000 network on my boat).

crbn60, 2022-05-08:
I would also like to help test. I have a non-critical setup in a mobile
workshop consisting of a REC Q BMS, Cerbo GX, MultiPlus, and SmartSolar
VE.can.

I'm using the 250 Kbit bus for the REC, I didn't know there was a reason to
use another.

mpvader, 2022-05-08:
Nice at both. That would be great. In a few weeks; or earlier maybe; I hope to have a Venus OS beta version with this included.

thats the best time to test

kwindrem, 2022-05-08:
Have there been changes to the main Venus OS GUI overview pages? If so, can someone point me to the documentation or at least the affected qml files so I can work this into GuiMods. It's been a request of many to see as separate tile for alternators or at least to show them in a DC System details page (which I haven't written yet).

Device menu pages aren't generally affected by GuiMods so that's not as much of an interest to me but do look at them for dbus parameters I'd need to work into the overviews.

system calc changes are also of interest since I get much of the data I display from those parameters.

nmbath, 2022-05-08:
@kwindrem, This [link](https://github.com/victronenergy/venus/wiki/dbus#alternator-and-dcsource) will give you the dbus paths that are generated. If some screen shots will help I can provide some of those. No changes have yet been made to systemcalc, so the generated current is included within the DC Load at present.

kwindrem, 2022-05-08:
Interesting: no DC power parameter? I thought Victron was moving away from current parameters. No biggie, it's a simple calculation.

nmbath, 2022-05-08:
@kwindrem, Yes there is power, see this dubs-spy capture:

```
om.victronenergy.alternator.socketcan_can0_vi0_uc67979
Connected                                                                                                    1
CustomName                                                                                           Wakespeed
Dc/0/Current                                                                                             636.7
Dc/0/Power                                                                                                8404

```

So good catch will get the link updated

thomasonw, 2022-05-09:
All,
We at Wakespeed are VERY excited about this milestone!   There are a couple of details for folks to know:

1) To get 'power' readings one will need to install a current shunt in the Alternator B+ output and connect the WS500 shunt sensing wires to it.  In addition, one will need to configure the 'Shunt-at-bat' as false indicating the shunt is located not at the battery.  AND one will want to use a BMS which is able to supply battery current via the CAN (Lynx is a great choice!)

2) We have made some changes in WS500 firmware to support the Victron release.  We have it in Alpha phase now, and will time its release into Beta to coincide with Victron's beta release.  For this special time we will post the 2.5.0 beta to the Wakespeed website.

3) Make sure to use a cross-over RJ45 cable, as the pinouts are different between Wakespeed and Victron.   Also make sure to use proper terminators  (hint:  Victron cables and terminators tend to be Blue, while Wakespeed's tends to be black).

millijuna, 2022-05-09:
@thomasonw 

In regards to your point 1 above, does that mean a high-side shunt on the B+ output from the alternator? I'm still working on the final installation on my setup, and in my design, I can have either a low side shunt between the engine and my house bank, or could put the same shunt in on the high-side between the alternator and house bank. Either one would only measure the alternator output (due to having a separate starter battery, and isolated DC:DC charger for said starter battery)

nmbath, 2022-05-09:
@millijuna,

It can go in either. In the setup I have it is in the positive side on the B+ output from each alternator. The reason for this is that the alternator is case grounded so the engine is the negative path. If I therefore put it on the negative side it will also pickup engine start currents, and the primary alternator on the engine, which is also case grounded. So while your DC/DC maybe isolated, you need to look at the negative/ground path of the alternator.

thomasonw, 2022-05-10:
@millijuna   Yes, the Positive output post of the alternator, so that the alternators amperage may be sensed.  If you have (not too common) an isolated ground alternator, then you can also place the shunt on the gnd wire, but as nmbath pointed out -- most alts are not isolated, and as such installing a shunt in the ground wire will likely pick up other current producers and/or consumers.

millijuna, 2022-05-20:
Happy to report that my system (put together in the last few days) is working nicely with the latest Victron public beta and version 2.4.3 of the WS500 firmware. From the Cerbo I can see alternator amps, alternator voltage (same as my bus voltage, but good to have), alternator speed (and engine tach), and alternator temperature. About the only thing missing is the field drive number, but I presume that is coming with version 2.5.0 of the wakespeed firmware.

martinbosma, 2022-05-31:
Currently Venus logs alternator voltage, current and temperature to VRM. 
Additionally also engine rpm and field drive % should be logged.
@izak Can you make VRM attributes for the following items and make sure a graph is created on the advanced page?
- /Engine/Speed - Engine RPM
- /FieldDrive - Current limit in %

I think we also should add those items to modbus-tcp. See the [wiki](https://github.com/victronenergy/venus/wiki/dbus#alternator-and-dcsource) for all the alternator paths.

thomasonw, 2022-06-01:
All,

Today we posted Wakespeed Beta firmware v2.5.0 to our website: [http://www.wakespeed.com/technical.html](http://www.wakespeed.com/technical.html)

It has several features, but one of the key ones is adjustments we made in conjunction with Victron as they extended the Venus OS to be 'Wakespeed Aware'.

A couple of quick notes (We should have release notes up in a few days)

- This is Beta, please give feedback, but do not send out to customers that will be driving away!
- The Cerbo uses N2K style messages for its integration, you will need to leave the 'Enable N2K' bit on (See $CCN command; N2K support is enabled by default)
- Connect the Wakespeed WS500 to the VE.can port, not the VE.bms port.  Make sure to use a cross-over RJ45 cable, and each companies terminators (the RJ45 pinouts are not the same)
- Victron makes use of charging devices current.  In most installs, the WS500 current shunt is placed at the battery.   However, if you are using a CAN connected BMS (ala, the Lynx) battery current is supplied by the CAN connection -- leaving the WS500 current sensing ckt unused.  You can place a current shunt on the alternator (Positive line is fine), connect up the WS500 shunt sensing lines, and then tell the regulator the current shunt is active and not on the battery (See $CCN command to tell the reg the shunt is 'not at the battery', as well as make sure the 'ignore sensor' field in the $SCA command does not have the current shunt disabled)

This has been a long time coming, and I know customers from both Victron and Wakespeed have asked for a while as well.  We are very pleased to have this release available for our customers and look forward to hearing how it can be helpful, as well as any issues / suggestions that you might find!

millijuna, 2022-06-02:
@thomasonw Just a heads up that your technical page references a 2.5.0 release notes PDF, but doesn't actually provide a link to it. I'll probably be bumping mine up this weekend.

nmbath, 2022-06-12:
While the data is being feed to VRM, is there any work required to allow the additional fields being feed to VRM included in the advanced graphing section? I only see Alternator Temperature, Current and Voltage. The data for Speed, FieldDrive, State and ErrorCode are also being feed into VRM. Even in Custom Widgets they additional data items are not available.

izak, 2022-06-12:
@nmbath the issue for adding the attributes to VRM is still open. A little more patience needed.

stevomyatt, 2022-06-24:
@thomasonw 

Some feedback on my current experience with a LynxBMS.

I can't get the local shunt current to display on VenusOS. There is a slight change between WS 2.4.3 and WS 2.5.0, I'll try to explain:

With 2.4.3, when CAN disconnected, local current shunt is displayed as both ALT and BAT current (AST;). When CAN is connected, both ALT and BAT current changes to Lynx BMS current. This current is displayed on VenusOS.

With 2.5.0, same as above, however when CAN is connected, current is not displayed on VenusOS, only dashes.

WS HW version 2 (printer USB socket). VenusOS 2.90~14.

[image]
[image]
[image]
[Lynx.Lithium.configFile_Local Shunt.txt](https://github.com/victronenergy/venus/files/8976149/Lynx.Lithium.configFile_Local.Shunt.txt)

nmbath, 2022-06-24:
@stevomyatt did you also just post this to the community forum? If so there is a response there. You need to check the shunt location physically and the configuration within the WS500 for where the shunt is located. If the WS500 believes the shunt is at the battery it can not report alternator current. You need to locate the shunt at the alternator and configure the regulator accordingly,

stevomyatt, 2022-06-26:
@nmbath yes, also posted to Victron community, not sure where is most applicable. As mentioned, shunt is wired to alternator positive and connected to WS sense wires. Both shunt at batt = false & ignore sensors = 0 is set correctly.

stevomyatt, 2022-07-03:
@thomasonw @nmbath FYI all appears to be working now that I have tested with the alternator running. Alternator current coming through after a short delay, presumably to ensure the shunt is working correctly.
Victron integration is fantastic, peace of mind at the fingertips that the system is operating correctly, being able to monitor field output and alternator temperature is very beneficial!

mpvader, 2022-08-17:
Hi @stevomyatt thank you for your help in testing! I'm going to close this issue now.

The one open point, adding alternator data into the html5-app is part of the ongoing html5-app redesign. We won't add it to todays version.

closed 2022-08-17 by mpvader

scottwb96, 2022-08-23:
@stevomyatt did you make any changes from when you weren't getting the data displayed to "appears to be working now"? 

I am running v2.5 beta. Shunts wired to B+ side of alt. Sense wires connected per the Wakespeed diagram (purple sense on alt side). Shunt at battery "false". Ignore sensors set to "0" as well. 
All fields except the 2 circled are populating..

[image]

[image]
.

Tuckie, 2022-08-23:
Can I ask if the ws3000r is included in these changes?

martinbosma, 2022-08-24:
> Can I ask if the ws3000r is included in these changes?

No, only the WS500 is supported for now.

stevomyatt, 2022-08-24:
@scottwb96 I have to apply some throttle for a few seconds to get some current flowing from the alternator before it shows up (my alternator does not charge at idle, RPMs are not quite high enough - I may look to reduce pulley diameter to get some charging at idle in the future). As I understand it the WS has to see some current through the shunt before registering it to make sure the shunt is actually working and not just failed low. Other things to check I guess is that the field output from the WS is energised and the alternator is actually producing current.

nmbath, 2022-08-24:
@scottwb96 This should really be handled in the Victron community forums. Something is looking very strange. If the WS500 has been powered and the engine is running you should have the other fields populated. Your screen shot of the WS500 screen is showing no real data populated.  While the shunt will not report any current until it has seen some flow, it should be reporting voltage, status, field drive, speed, rpm, etc. It would be worth checking your CAN wiring and providing more detail.about your setup and connections.

scottwb96, 2022-08-24:
I am definitely getting current from the alt's. I can see that flowing through the smartshunt and onto my Axiom display (when moving).

I will check again when I am back at the boat tomorrow at a higher RPM and see if they populate then. I do get data populated in all the other categories listed below...
Output (1) = Voltage
Output (2) = --
Output (3) = --

Temperature = varies but data is populated
State = Slave (if I remember correctly)
Field Drive = % is populated
Speed = data populated
Engine Speed = data populated

scottwb96, 2022-08-25:
Went down to the boat today to check the higher RPM's amps. Still no change on the 2 empty boxes.... hmmm..
I also checked the B+ output with an amp meter, definitely getting current flowing into the batteries (through the B+ connected shunts)...

Thoughts?!?!? :)

[image]
[image]

[image]
[image]

.

nmbath, 2022-08-25:
Please supply a copy of the ASCII output from connecting to the serial interface. Without that it is difficult to know. We need several lines with of AST as well as the other lines which are delivered.

kwindrem, 2022-10-11:
Finally getting around to a DC System details page. 

I would display a custom name for each device if provided. If there isn’t a custom name, what do you recommend I show in the name field?  I expect the order of devices may change with each system boot. I’m not sure I can produce a stable default name. That is, what was Alternator 1 may become Alternator 2.

Each row would contain the device’s name if it exists. ??? if does notI expect the order of devices may change with each system boot. I’m not sure I can produce a stable default name. That is, what was Alternator 1 may become Alternator 2.

Device type (alternator, fuel cell, DC source DC load
Device’s power
Device’s direction: Supplying / Consuming
State: off, bulk, absorption, float, etc
ALARM - if any alarms are present and active)
	or display a specific alarm and display them in scrolling text for more than one
		high voltage
		low voltage
		hight temp
		low temp
ALARM might replace state if I can’t fit both columns

I could show a summary for all PV Chargers in stead of a row for each PV charger device since there’s already a details page for PV chargers.

I’d include inverter/chargers or inverters - a row for each
Do Victron chargers provide information ?

As of v2.90~8, I find the following devices that I think should be included:

 	DBusService.DBUS_SERVICE_FUELCELL:
	DBusService.DBUS_SERVICE_DCSOURCE:
	DBusService.DBUS_SERVICE_DCLOAD:
	DBusService.DBUS_SERVICE_DCSYSTEM:

I see the following com.victronenergy.system parameters:

	Dc/Charger/Power
	Dc/FuelCell/Power
	Dc/System/Power - this is the total power displayed in the DC System tile I think.
		It would display at the top of the details page along with a power gauge

Are the Charger and FuelCess system parameters of any value?

Anything missing?

The document you reference has no dbus service (com.victronenergy.????) for alternator, fuel cell, DC source DC system.
I will probably need these for gathering device parameters

Not sure how to deal with additional batteries. Any ideas?

Don’t see parameters for DC/DC converters. I think Victron Orion DC/DC converters and charterers don’t  report to GX?

—
Kevin

natecostello, 2023-03-14:
I know this issue is closed, but I'm seeing something strange that may be a bug.  I recently reconfigured my system to use a dedicated alternator shunt, and to rely on BMS reported current over CAN for battery current.  After that change my Cerbo reports ≈2x KW when the alternator is charging.  The battery KW is correct.  It seems like this might be a sign error, but I don't think it is an error on the wakespeed side.  When I monitor the CANbus, wakespeed is reporting the current it is measuring on its dedicated shunt as positive, which I believe is correct.

My DC system, enabled via the "has DC system" option on the CERBO, seems to be taking the battery current, and adding the wakespeed current to calculate DC power, hence the ≈2X.  If I'm missing a setting somewhere I'm sorry, but I haven't found anything obvious from the documentation, or answers at the Victron Community forum or offgridmarine (dealer for vic and wake) forum.

kwindrem, 2023-03-14:
There are a number of errors and omissions in the calculated DC System value in systemcalc. In the active code, alternator power is not included. A commented out version subtracts alternator power. My experimentation suggests alternator power should be added. I don't think the active code accounts for a 2x error though.

nmbath, 2023-03-14:
@natecostello You really are better off asking questions on the community forum, rather than here. As Kevin has pointed out the DC load is calculated and can be misleading. It is based on amps left over. If someone does not have a WS alternator then amps should be excluded from the calc, and can be argued if their is an WS it should be included (or visa versa depending on how you look at it). The simplest is to put a smart shunt in the DC loads negative path and configure the shunt as a DC systems shunt. The system will then not calculate the amps, but use that shunt reading. Whatever you connect to the load side is your reported dc load. Works like a charm and no ambiguity or calculation issues.