victronenergy/venus #781: invalid json in payload on device alternation

woodz-, 2021-02-09:
In your Readme.md you state that

    Topic: N/e0ff50a097c0/pvinverter/20/Ac/Power
    Payload: {"value": 936}

is the format on how transmission works by your dbus-mqtt module.

I assume it's a common standard and there should be a kind of a reliance that payload data is always valid json by no exception, right?

Web view of Venus OS (likely dependent on the dbus implementation) shows somehow curios behavior. Form time to time it happens that devices disappear.

In my specific case it's the external pv-inverter, a `STP 8000TL-20`, which disappears when solar power (and the daylight) goes down. The time interval when it disappears is variant. Sometimes it happens 10 minutes after reaching zero power, sometimes it takes a couple of hours and sometimes it keeps appearing overnight (doesn't disappear at all).

It reappears, when daylight comes back and solar power rises within the position of the sun. The effect during those disappearing and reappearing phases is always the same. In those phases invalid payload data gets transmitted via mqtt.

I am working on a python script and I am watching the topic

    "N/<individual_portal_id>/pvinverter/20/Ac/+/Power"

which particularly delivers the power of the three phases L1, L2, L3.

When it comes to one of the transition periods mentioned above, I receive an empty octet string: `b''` which definitively leads to a json parser error on position `0`

As a consequence it is required to validate input data - the mqtt json payload.

Is this expected behavior?

jhofstee, 2021-06-17:
From the looks of it, it is intentional, see https://github.com/victronenergy/dbus-mqtt/blob/master/dbus_mqtt.py#L89. Just scanning the code though, I haven't checked it.

mpvader, 2021-08-17:
this is as its intended, even if a bit weird.

closed 2021-08-17 by mpvader