victronenergy/venus #783: Clear up selecting a bmv while there is a bms

mpvader, 2021-02-15:
For a system with a BMV and a managed battery, we currently support the user selecting the bmv as the main  battery monitor in the settings -> system setup menu.

And then what systemcalc does, is that it still finds the BMS and uses it for BOL control (CCL, CVL, DCL). In case of multiple BMS-es it will use the one with the lowest device instance.

That was made a while back, to allow users using a BMV in combination with their managed battery.

But its not documented anywhere publicly and also the gui doesn’t show it to the user. Ie a hidden feature.

Next step: either finish this (show it in the gui, document it etc) or remove this feature and make it such that if the user selects the BMV, the whole BMS is ignored. Which is nice and simple.

izak, 2021-04-06:
I want to argue for the _nice and simple_ approach: removing the hidden feature.

Since Venus 2.30, hub4control (which controls whether the battery can discharge and how fast) always uses the selected battery service. This was part of making the BMS selection more sane where there are multiple managed batteries.

In short: In an ESS system, selecting the BMV already makes it ignore the BMS, which affects whether the battery can discharge and how fast it can discharge.

Systemcalc however still uses the BMS with the lowest deviceinstance as information source for the Multi and Solar Charger's charge voltages and current limits.

The correct thing would be to make things consistent.

---

We will however break things for a few people. The reason why we allowed this split is because at the time there were already existing customers using a BMV because the managed battery in that system didn't track SOC very well.

This is due to poor current tracking at low power levels, usually batteries that use a single Hal-effect sensor. The effect becomes worse when there are multiple modules in the battery, each with their own sensor.

Removing the ability to use another battery measurement will affect:

 * SOC: There can be substantial SOC drift with an inaccurate current measurement.
 * SCS: Writing back the battery current to the solar chargers is affected if the measurement is inaccurate.

A non-exhaustive check on Grafana indicates around 30 sites (with a BYD, BlueNova, or Generic CAN-BMS) that use a BMV as the selected battery monitor even though a managed battery is in the system.

To deal with that, I will create another issue for finding an alternative way to deal with poor SOC tracking and using an "auxiliary" battery monitor.

izak, 2021-06-10:
This is my solution for systems that use a BMV or smartshunt to get better SOC numbers.

1. It specifically looks for a BMV/Smartshunt in the system. If there is more than one, the one with the lowest device instance is used.
2. There must be at least one other battery in the system.
3. The BMS/Smartshunt must not be the selected battery service.
4. The option is only available with DVCC on, because without DVCC you can just select the managed battery as battery service.

Under these conditions, a switch becomes available to use the BMV/SmartShunt for SOC. Poor font in picture due to running on PC.

[image]

The label in the GUI needs a better name. For one, it should probably use "Managed Battery" instead of BMS, but that makes the label fairly long.

mpvader, 2021-08-17:
Done per v2.80, closing.

closed 2021-08-17 by mpvader