victronenergy/venus #784: dvcc: Force settings for pylontech (or update documentation).

izak, 2021-02-16:


izak, 2021-03-29:
Rather than force the settings, which will affect a very large number of users, I updated the documentation.

mpvader, 2021-03-29:
Ok but lets keep the issue open still; we need to fix this somehow someday.

And which documentation did you update? Don’t forget to check the gx manual.

izak, 2021-06-04:
Some statistics that may be helpful.

Total of 8115 Pylontech batteries on VRM.

| Total installations | DVCC on | DVCC off | % |
|---|---|---|---|
| All installations | 7083 | 1032 | 12% |
| Venus OS >= 2.50 | 6882 | 940 | 12% |
| Venus OS >= 2.60 | 6168 | 825 | 12% |

Roughly 12% of Pylontech installations don't use DVCC.

mpvader, 2021-06-04:
very helpful! lets force it on. and then make sure to do it at the start of a beta series that takes a while. Ie the complete opposite of what would happen if we'd make that change now in the latest v2.70 release candidate.

mpvader, 2021-12-15:
closing. doc updates going into https://github.com/victronenergy/venus/issues/882

closed 2021-12-15 by mpvader