victronenergy/venus #787: MQTT Topic incorrect escaping of quotes in JSON

Knobee, 2021-02-24:
Good day, all!

Discovered in v2.62, remains in v2.63

I've just deployed my first functional MQTT broker and have discovered that, buried deep in the tree is a topic:

```
N/+/system/0/AvailableBatteryServices
```

that, unlike it's neighbors, has an extra set of escaping backslashes throughout the message:

```
{"value": "{\"default\": \"Automatic\", \"com.victronenergy.battery/289\": \"BMV-712 Smart on USB\", \"com.victronenergy.vebus/288\": \"MultiPlus 12/3000/120-50 120V on VE.Bus\", \"nobattery\": \"No battery monitor\"}"}
```

The neighboring topic:

```
N/+/system/0/AvailableTemperatureServices
```

contains the correct:

```
{"value": {"default": "Automatic", "com.victronenergy.vebus/288/Dc/0/Temperature": "MultiPlus 12/3000/120-50 120V on VE.Bus", "nosensor": "No sensor"}}
```

mpvader, 2021-02-24:
Thank you for reporting @Knobee , we’ll check.

@izak when testing the changes, make sure the HTML5 app still works.

jhofstee, 2021-02-24:
From the looks of it, it is correct. It is json, within json..

jhofstee, 2021-02-24:
```
 'AvailableBatteryServices': '{"default": "Automatic", "com.victronenergy.battery/256": "BMV-712 Smart on VE.Direct", "com.victronenergy.vebus/257": "MultiPlus 24/3000/70-16 on VE.Bus", "nobattery": "No battery monitor"}',
 'AvailableTemperatureServices': {'com.victronenergy.battery/256/Dc/0/Temperature': 'BMV-712 Smart on VE.Direct',
                                  'com.victronenergy.vebus/257/Dc/0/Temperature': 'MultiPlus 24/3000/70-16 on VE.Bus',
                                  'default': 'Automatic',
                                  'nosensor': 'No sensor'},
```

Note that AvailableBatteryServices is a string, with json in it! AvailableTemperatureServices seems to be a dict. So at least the escaping doesn't seem to be the issue (as the subject says)

izak, 2021-02-24:
> Note that AvailableBatteryServices is a string, with json in it!

Indeed. I do not know if this is the main reason, because this happened before my time, but I suspect it was done this way because of the difficulty of getting complex types marshalled into QML in the Gui. Using a json string makes it easy, you can just `JSON.parse()` it. Fixing it will therefore require changes to the GUI as well.

jhofstee, 2021-02-24:
I sincerily doubt it is for the qml. It should happily deal with a Dict.

jhofstee, 2021-02-24:
localsettings is a better suspect, since it only knows about ints, doubles and string. But I don't see how that is related with this.

Knobee, 2021-02-24:
So, the expectation is that I'd be required to know the difference between, and parse:

`N/+/system/0/AvailableBatteryServices` and
`N/+/system/0/AvailableTemperatureServices`

in different ways?

To be honest, I'm doing neither (yet).  I found the resulting values to be displayed differently in the app that I'm using to create panels from the Victron created MQTT data and believed it to be an anomaly.  I'm going to be installing 2x Victron 200Ah LiFePO4 batteries as soon as they are shipped to me, so I do believe I'll soon be in the throes of attempting to use these entries soon.

I agree that the strings `AvailableBatteryServices` and `AvailableBatteries` do provide **legal** JSON, it's the issue with the (lack of) parallelism between these entries and the other data in the same branch of the MQTT tree.  Using nothing but eyeballs, it appears that the two entries mentioned here are the only ones that contain strings encapsulating JSON in the entire tree.

If this is a problem with too complex a solution to make it worthwhile, I have no issue with it being marked 'no fix', but it seems that the two "strings containing JSON" mentioned here should be exposed to the client as the contents of the string and not the string itself.

[Perhaps I'm misunderstanding the discussion here and everyone is in fierce agreement that there is an issue] 🥇

_The fact that I'm able to provide feedback and get developer response on issues like this is what sets Victron apart from the other vendors in this arena._

Knobee, 2021-02-24:
[image]

output provided by [MQTT-Explorer](https://mqtt-explorer.com/)

jhofstee, 2021-02-24:
> So, the expectation is that I'd be required to know the difference between, and parse:
> 
> `N/+/system/0/AvailableBatteryServices` and
> `N/+/system/0/AvailableTemperatureServices`
> 
> in different ways?

Yes, that how it seems to work as is. Don't ask me why, since I simply don't know. Likely @izak can dig that up if he has time, but for now that is the way it is.

mpvader, 2021-03-08:
hi @Knobee we agree its not perfect, but it is what it is: its used in several places in our code; and maybe also other customers code; so too complicated to fix.

I'll close this issue.

closed 2021-03-08 by mpvader