victronenergy/venus #788: mqtt-rpc: Add remote firmware update feature for GX device.

izak, 2021-02-24:
TODO
- [ ] List GX device in device list
- [ ] Call `check-updates.sh -swu URL` with returned url if type is `swu`.
- [ ] Document in README.txt

mpvader, 2021-09-06:
This was waiting for it to be deployed on VRM by the VRM developers. But as discussed just now, we can already merge it now; its better.

mpvader, 2021-12-15:
@izak, I did a quick test, and doesn't seem to work?

See the error 1342 in the screenshot.

Also, would be nice to have the GX Device at the top; just like in the Device List on VRM.

[image]

mpvader, 2022-01-27:
Implementation is complete in Venus v2.80. Whats missing is implementation in VRM, which is documented here: 

https://github.com/victronenergy/vrm-portal/issues/228

I'll discuss with Barbara.

I'm leaving this issue open, to not forget to follow up if need be.

mpvader, 2022-08-16:
status, as discussed just now [https://victrondevelopment.slack.com/archives/CJJ24D30X/p1660658817546049](here): this feature is finished in both Venus OS, as well as VRM. But not yet deployed to production of VRM, which will do only a while *after* releasing v2.90 officially. 

More on that in the linked thread.

izak, 2022-12-07:
I looked in the wrong place. This is indeed not deployed yet. Reopening.

mpvader, 2023-03-01:
@izak can this now be closed? Iirc this feature is now deployed?

izak, 2023-03-01:
Yes. Closing.

closed 2023-03-01 by izak