victronenergy/venus #789: D-Bus: send multiple properties changed from root?

mpvader, 2021-02-26:
An idea to reduce D-Bus traffic:

Now, for example a solar charger, sends multiple PropertiesChanged signals every second, one for each D-Bus path.

The idea is to send only one every (for example) second, from root, with a Dict with all paths that have an updated value.

Thiemo modified vedirect-interface for this; Izak knows branches. And also Thiemo fixed the QItems used by the GUI, to cope with this new method.

This is in a proof of concept phase, to see how much D-Bus traffic and thus CPU usage we can reduce with this. My expectations are quite high.

Next step is for Izak to fixup the python processes to cope with this, and then run this on a busy system to see the effect.

mpvader, 2021-03-07:
branch is here:
https://github.com/victronenergy/vedirect_interface/commits/thiemo/WIP_dict_updates

izak, 2021-05-10:
Some results from testing.

## Test environment
1. Venus 2.65 base system
2. vedirect-interface replaced with a binary that sends dict updates on root
3. systemcalc and vrmlogger patched to use velib_python that understands dict updates
4. gui recompiled with new velib so it understands dict updates
5. Rest of the system left as is (should have minimal effect on CPU usage).

This was tested vs an unmodified setup.

## First test: Test effect on CPU use on a working system
System has:
1. CCGX
2. 2 x VE.Direct solar chargers
3. 1 x Multiplus
4. 1 x BMV 700
5. System was set up to feed solar power into the grid, so that there would be constant power changes to communicate.

### Unoptimised system
CPU use:
* systemcalc: +-10%
* vrmlogger: +-10%
* dbus-daemon: +-6%
* vedirect-dbus: +-2%

CPU idle: +-30%

### Optimised system
CPU use:
* systemcalc: +-8%-9%
* vrmlogger: +-8%-9%
* dbus-daemon: +-5%
* vedirect-dbus: 3%-5% (slight increase)

CPU idle: +-40% (handsome improvement)

## Second test: DBus throughput on a heavily loaded system
System has:
1. CCGX
2. 2 x VE.Direct solar chargers
3. 1 x Multiplus
4. 1 x BMV 700
5. System was set up to feed solar power into the grid, so that there would be constant power changes to communicate.
6. An additional 5 simulated solar chargers that change voltage/power/current values once a second.
7. dbus-statistics look specifically at PropertiesChanged signals and ignores SetValue/GetValue calls.
8. Statistics were assessed using wireshark and dbus-monitor's --pcap option.

### Unoptimised system
* Dbus RTT: > 5000ms (round trip time was over 5 seconds)
* Dbus traffic: 7400 bytes/sec
* Messages per second: 37
* Average packet size: 197 bytes
* dbus-daemon CPU load: 10%-15%
* Watchdog rebooted the system as load average exceeded 10.

### Optimised system
* Dbus RTT: > 100ms-200ms
* Dbus traffic: 5900 bytes/sec
* Messages per second: 15
* Average packet size: 388 bytes
* dbus-daemon CPU load: 5%
* Load average was still very high (due to simulated solarcharger processes).

### Very heavily loaded optimised system
Another two simulated solarchargers were added, for a total of 7 simulated and 2 real solarchargers (plus a BMV).
* Dbus RTT: 2000ms
* Dbus traffic: 6400bytes/sec
* Messages per second: 17
* Average packet size: 376 bytes
* dbus-daemon CPU load: 5%
* Watchdog rebooted due to very high load, even though the dbus-daemon remained fairly responsive

## Conclusion
* Overall CPU idle time was increased (good)
* dbus is no longer the bottle-neck
* Messages/second reduced to a little less than half (good)
* Message size roughly double what it used to be
* dbus-daemon CPU use remains low (good)

## Other observations
* vrmlogger's rtt implementation is skewed under heavy load, because the load retarts the event loop and increases the measurement.
* There are still many gains to be had in dbus-cgwacs and dbus-fronius, especially in a three-phase system where these two create quite a bit of traffic, and especially where multiple PV-inverters are present.

izak, 2021-09-06:
@mpvader I remember now what's holding back improvements on the QT projects.

There is a branch (`thiemo/WIP_Item_Dict_Leafs`) that implements the sending of multiple values in a single message, but only for `C` projects (this was used for vedirect-interface when doing this evaluation). A C++ implementation is needed.

That same branch has code for receiving multiple values in a single message. This does have a C++ version and was used during testing in the GUI.

`velib_python` has a `dict_updates` branch that is fully up to date (python3 ready). It implements a context manager that can be used with a `with` statement, and it sends everything that happens inside that block together in one message.

What is needed is a C++ implementation that allows grouping multiple values in one `PropertiesChanged` message.