victronenergy/venus #791: vrmlogger: better handle D-Bus issues

mpvader, 2021-03-05:
While dbus-systemcalc seems to cope fine, vrmlogger gets into a problem; end result being that it hasn't sent any data to VRM for over an hour.

``` 
2021-03-05 08:59:07.875783500 MainThread-dbusmonitor: Found: com.victronenergy.solarcharger.socketcan_can0_vi17_uc79934, scanning and storing items
2021-03-05 08:59:07.879334500 MainThread-dbusmonitor: Ignoring com.victronenergy.solarcharger.socketcan_can0_vi17_uc79934 because of error while scanning:
2021-03-05 08:59:07.879533500 Traceback (most recent call last):
2021-03-05 08:59:07.879674500   File "/opt/victronenergy/vrmlogger/./ext/velib_python/dbusmonitor.py", line 171, in scan_dbus_service
2021-03-05 08:59:07.879912500     return self.scan_dbus_service_inner(serviceName)
2021-03-05 08:59:07.879992500   File "/opt/victronenergy/vrmlogger/./ext/velib_python/dbusmonitor.py", line 195, in scan_dbus_service_inner
2021-03-05 08:59:07.880131500     serviceId = self.dbusConn.get_name_owner(serviceName)
2021-03-05 08:59:07.880189500   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 361, in get_name_owner
2021-03-05 08:59:07.880313500     's', (bus_name,), **keywords)
2021-03-05 08:59:07.880374500   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
2021-03-05 08:59:07.880489500     message, timeout)
2021-03-05 08:59:07.880667500 DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'com.victronenergy.solarcharger.socketcan_can0_vi17_uc79934': no such name
```

Here is the full log file:
[vrmlogger.log](https://github.com/victronenergy/venus/files/6089395/vrmlogger.log)

Underlying root cause are issues on the canbus; but regardless of proper or non-proper functioning canbus, vrmlogger needs to be resilient.

First step, Izak to have a look and discuss how hard/easy it is to fix this to determine priorities. If difficult to fix, then not too urgent.

izak, 2021-03-26:
It is not clear to me how this exception could stop vrmlogger from logging. The scanning of a dbus service is wrapped in a bare try/except which prevents any of these exceptions from crashing a gevent timeout loop elsewhere (added using `gevent.timeout_add()`). The error also happens on a NameOwnerChange, so is not part of such a timed event loop.

If it really stops logging, the problem is much deeper than an exception that is not being handled.

Next step would be to attempt to reproduce.

mpvader, 2021-03-26:
looking at the long log file (see the link), it looks to me that its all the time busy processing new (non-responsive) canbus services that are continuously coming and going, and somehow never gets to sending a log.

izak, 2021-04-15:
Finally observed this in action at what I assume is the same site. The problem is that vrmlogger is scanning the service (reading all it's data with a call to GetValue) directly after it comes up, and while it is doing this it cannot do anything else (single-threaded event loop type setup). Because the solarchargers come and go repeatedly, vrmlogger ends up spending all of its time scanning new services and none of it logging.

mpvader, 2021-08-17:
Discussed it with Izak; there is nothing trivial that can be done about it. Also, the iirc, dbus-vecan has been changed, its timeout on the can communication with the solar charger has been increased or improved. So this is now less likely to happen.

Closing; and if necessary this will come back by itself.

closed 2021-08-17 by mpvader