victronenergy/venus #794: html5-app: battery status has a 0 postfix

mpvader, 2021-03-07:
[image]

AlexImb, 2021-03-07:
@mpvader This apparently happens when the reported SOC value is 0. I changed it so it is formatted as a percentage as well:

[image]

mpvader, 2021-03-07:
Ah thanks. Looking at it this closely, I do see something else: the formatting of “charging 0%” is a bit ambiguous. Could mean that its charging with 0% power.

instead of being at 0% state of charge, and being charged. Charge charging / idle / discharging belongs with the Amps and Watts; could you try move it there?

AlexImb, 2021-03-07:
I made a new minor release (1.5.4) for the fix I mentioned above and included it to the Venus todo list. I think that's still a good improvement to prevent in from looking like a postfix/bug. 

About moving. On the right side the "time to go" is also shown. Should that also go to the left or stay together with the SoC on the right.

Current implementation:

[image]

mpvader, 2021-03-07:
Yes, I noticed the bugfix on the todo, thats good lets leave it there. Thanks for the quick fix!

And the time to go.. like this, with the time to go in between the Discharging and the SOC percentage, it looks better. Without the time to go, as in a few screenshots up, it doesn't.

The time to go belongs with both the SOC and the discharging/charging/idle. ;o).

So.. I don't have a good solution now; knowing this situation it isn't good enough to just move "discharging" to the left.

mpvader, 2021-03-10:
Closing this - the bug was fixed and is in meta-victronenergy/master now for v2.70 as well as in v2.65~5.

The part of the discussion about moving the text a bit will resurface automatically some day in case important enough.

closed 2021-03-10 by mpvader