victronenergy/venus #795: Beaglebone Cape Firmware

silverchris, 2021-03-07:
Hi All,
I was wondering if there was a release of the firmware for the PIC18F24K2 on the beagle bone cape.
So I can replace/reflash the microcontroller.
Thanks!

mpvader, 2021-03-07:
Hi sorry but no; we don’t provide that as a hex file; this is about a Venus GX? Most likely still under warrant, not?

silverchris, 2021-03-07:
I am actually currently working on a project to build an integrated power supply + can hat for the raspberry pi.

I was hoping that since the schematics for the beaglebone cape were published, that the firmware for the mk3 interface would also be available, as it would allow me to create a very compact solution for when I purchase a victron inverter to go with the rest of my setup

mpvader, 2021-03-07:
Ok; could you send me an email?

closed 2021-03-07 by mpvader

silverchris, 2021-03-07:
> 
> 
> Ok; could you send me an email?

Sure, could I get your email? Or you could email me. silverchris@gmail.com