victronenergy/venus #801: IEEE 802.1x

mbr1989, 2021-03-16:
Why is there no IEEE 802.1x support ?

mpvader, 2021-03-16:
Probably because the number of people that work with network based access control is too small to warrant the effort. You're the first to ever ask.

The good news, as you probably know large parts of Venus OS are open-source. Especially the network part is. And root access is available as well.

So you'll (a) possibly be able to modify a running GX device to make that work; or (b) modify sources, build, compile, test and eventually send us a series of patches to add it.

For a more commercial answer, please contact your Victron dealer.

I'm closing this issue now, any further discussion please use the Modifications section on Victron Community or the mailinglist.

closed 2021-03-16 by mpvader

mbr1989, 2021-03-16:
... security is overrated

mbr1989, 2021-03-16:
"eventually send us a series of patches to add it" 

aha ... customer product development ... YEAHHHH

mpvader, 2021-03-16:
politely asking questions, suggesting ideas and providing context is overrated also.

mbr1989, 2021-03-16:
I'm not employed here as a polite questioner at Vicron.
Context here is security in electrical energy systems.

kogir, 2021-03-16:
The serial links that Venus devices use to communicate with and control the other Victron equipment are unencrypted and unauthenticated. 

If you can’t physically secure all the Victron equipment and their communication links, 802.1x will add zero additional protection. 

I work in security consulting and am having trouble imagining a threat model where 802.1x on Venus gear is relevant, and you’ve provided no details elaborating on how it’s uniquely suited to protect a Victron installation you can’t physically secure.

mbr1989, 2021-03-16:
> The serial links that Venus devices use to communicate with and control the other Victron equipment are unencrypted and unauthenticated.
> 
> If you can’t physically secure all the Victron equipment and their communication links, 802.1x will add zero additional protection.

But WiFi is here the first attack vector

mbr1989, 2021-03-16:
> Edit: Oh, you might mean for WiFi. Definitely far preferable to run a wired link wherever security is a concern.

Not possible due to mobile in-building application