victronenergy/venus #806: systemcalc: in an ESS system with multi off has issues

mpvader, 2021-03-29:
The grid meter doesn't show properly anymore:

[image]

while there is still data from it:

[image]

and a little bit later, it shows this, which is also strange and/or wrong:

[image]

most likely what above picture should do is: (a) show those loads connected directly to AC in, (b) still show critical loads, like two pictures up, (c) still show the ac input

mpvader, 2021-03-29:
also the VRM looks strange. In a slightly different way though. see screenshot below. And note how left and right are different.

Probably step 1 is to fix dbus-systemcalc to show the info in the proper way, issue is probably that it no longer sees an ESS Assistant.

And then step 2 is to get gui and vrm to use the data properly, if they don't already do that.

[image]

izak, 2021-03-30:
The reason why the grid meter value is not used when the Multi is off, is because the GUI decides which value to show in that box based on the active AC input, and when the Multi is off it doesn't know. So while the grid value is there, it doesn't know whether it can use it. This logic is [in the GUI][1].

[1]: https://github.com/victronenergy/gui/blob/30372937736226f045aacfadfefb88f00c0280cf/qml/HubData.qml#L26

izak, 2021-05-27:
Discussion of implementation to date.

1. Systemcalc publishes three new values corresponding to the active AC input. Therefore the GUI no longer needs its own logic and can simply use these values from systemcalc.
2. Systemcalc can make an intelligent decision based on the availability of a grid or genset service, and publish suitable information even when the Multi is off.
3. Fix the hubtype indication in systemcalc, so that the GUI doesn't change to a different overview (losing the Critical Loads block). This also fixes another irritation: Switching the Multi off causes the Tiles view to show "Hub-1", which is obviously wrong.

izak, 2021-08-13:
Fixed as of Venus v2.80~10.

closed 2021-08-13 by izak