victronenergy/venus #807: mfd app shows white screen both on MFD and also in a browser

mpvader, 2021-03-30:
seen by Martin.

martinbosma, 2021-04-09:
This happened with a v2.70 image with a change in the amount of free space on the rootfs. 
[image]

Running `resize2fs.sh` seems to resolve the issue.

`df` just after updating to 2.70 with resized partition; this leads to a MFD app page and weird remote console rending directly in the browser:
```
root@beaglebone:/var/www/venus# df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root               250444    236228         0 100% /
devtmpfs                247192         4    247188   0% /dev
tmpfs                   255896       208    255688   0% /run
tmpfs                   255896       232    255664   0% /var/volatile
/dev/mmcblk1p5          122835      4586    109075   4% /data
```
After running `resize2fs.sh`; this seems to resolve the issue:
```
root@beaglebone:/opt/victronenergy/swupdate-scripts# df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root              1261606    236919    965628  20% /
devtmpfs                247192         4    247188   0% /dev
tmpfs                   255896       208    255688   0% /run
tmpfs                   255896       232    255664   0% /var/volatile
/dev/mmcblk1p5          122835      4586    109075   4% /data
```

After this I'm unable to reproduce the issue. Not even when downgrading to v2.6x and upgrading to v2.70 again.

mpvader, 2021-04-09:
Spoke Mans on the phone, he knows the problem.

Its a problem in one of his branches; not a problem happening when building out of master.

He'll fix the issue in his branches before putting it onto the todo.

closing the issue

closed 2021-04-09 by mpvader