victronenergy/venus #811: Venus OS Large: fix node-red time always on UTC

mpvader, 2021-04-17:
The problem is that setting a time zone in the venus os gui has no effect on the time used in flows: node-red always operates in UTC.

now I didnt look into this yet, though from memory right now linux in Venus OS always operates in UTC, by design.

and local time is set, and used, in the gui only.

there are probably a few ways to go about this.

note that if indeed we’ll grt Grafana & Influx in Venus Os (Large), then those will likely suffer from the same issue.

dirkjanfaber, 2021-09-06:
The Node-RED determines the timezone by looking at `/etc/localtime`. This file ought to symlink to the local timezone. (See https://www.freedesktop.org/software/systemd/man/localtime.html)

In the case of Venus, this is set to `/usr/share/zoneinfo/Universal` instead of to the timezone that is set from the gui. A way to solve this would be to symlink `/etc/localtime` to `/data/conf/localtime`, which symlinks to the correct timezone again (defaulting to `/usr/share/zoneinfo/Universal`). 

Alternative would be to use the TZ variable when starting up the application. This can be done by adding something like:  
```
export TZ=$(dbus -y com.victronenergy.settings /Settings/System/TimeZone GetValue | tr -d "'")
```
to the `/usr/bin/node-red` script.

mpvader, 2021-09-06:
Nice. Go for the option in the startup script pls

mpvader, 2021-09-12:
This is fixed; will be available in large-image-22

closed 2021-09-12 by mpvader