victronenergy/venus #813: Using energy meter to monitor and display AC loads

ZacTat, 2021-04-22:
Hi all,

I am wondering if anyone has used a ET112 energy meter to monitor the output of a third party inverter? I haven't used an energy meter yet but I imagine it will be configured to display in VRM as an "AC Input" but in this use case id want it to show as an "AC Load" and log data as "Consumption" rather than being "To Grid". 

Has anyone done this? 

Cheers, Zac

ZacTat, 2021-04-22:
A bit more info about my setup, I have recently upgraded my gear to Victron, I have a BMV-712, MPPT and Cerbo GX. Really happy with how its running but I have an existing AC inverter that works well and I cant justify the extra cost to upgrade to a MultiPlus. It would be great if Victron could support monitoring third party inverter loads with the energy meters.

mpvader, 2021-12-15:
Hi ZacTat, we now do support usign energy meters to measure a load. 

Maybe you can use that for monitoring your inverter.

I'll close the issue: we won't be spending time to more detailed work on this I'm afraid.

closed 2021-12-15 by mpvader