victronenergy/venus #815: uboot sunxi: missing pythonnative depedency

mpvader, 2021-04-30:
Reported by Robert Berger.

I have more info about this if needed; need to go looking through emails for that.

```
diff --git a/meta-bsp/recipes-bsp/u-boot/u-boot-sunxi.inc b/meta-bsp/recipes-bsp/u-boot/u-boot-sunxi.inc
index c5b4accc..40aa70c0 100644
--- a/meta-bsp/recipes-bsp/u-boot/u-boot-sunxi.inc
+++ b/meta-bsp/recipes-bsp/u-boot/u-boot-sunxi.inc
@@ -9,3 +9,5 @@ SRC_URI[md5sum] = "018a2263934027a64b1ba23dfb49faeb"
 SRC_URI[sha256sum] = "34112392f1346e680a834c6eff77bbaca62d734988d30991d2941269d2d586c8"
 
 S = "${WORKDIR}/u-boot-${PV}"
+
+inherit pythonnative
```

jhofstee, 2021-06-15:
The plan is to drop the python dependency, since it depends only on python2, to cat some files.
https://github.com/victronenergy/meta-victronenergy/commit/4bd3f5b3c06f1b99e18ed2fba7d31d0218776848 should work for both python2 and python3 (since they are no longer needed).

jhofstee, 2022-05-21:
The python dependency has been removed.

closed 2022-05-21 by jhofstee