victronenergy/venus #820: VE.Bus: Add support for MultiPlus-II 2x120V

vejpasop, 2021-05-12:
Add support for the L2 split phase input and output.

Changes need to be made to:
- [ ] Multi firmware 
- [ ] vebus-interface
- [ ] GUI
- [ ] VRM

Vebus-interface:

A new D-bus path has been added `Ac/State/SplitPhaseL2Passthru` 
This path is only valid on a 2x120Vac device when the split phase L2 output is **NOT** disabled.
The L2 output is disabled when the Multi is configured as a true Multi phase setup, such as three phase where L2 consists of an actual Multi.
 `Ac/State/SplitPhaseL2Passthru` indicates if the L2 split phase input is connected to the L2 output.
When this path is valid. `Ac/NumberOfPhases` will always be 2

### Inverter mode
[image]
**D-Bus output**
[image]

### 120VAC single phase input mode
[image]
**D-Bus output**
[image]

### Split phase input mode
[image]
**D-Bus output**
[image]

vejpasop, 2021-05-17:
Included in mk2-dbus V3.21

jepefe, 2021-05-28:
I implemented the changes as discussed. gui branch: feat_overview_splitphase

L1 & L2 connected:
[image]

L1 connected | L2 disconnected:
[image]

mpvader, 2021-06-28:
@vejpasop & @jepefe : you're all done with this, right?

Which then leaves some work to be done by @izak, and thereafter perhaps also the overview on the VRM Portal.

Jos, please list which extra paths (if any?) need to be supported by Izak for that.

jepefe, 2021-06-28:
@mpvader Yes, and I see that gui changes are already included in gui v4.9.2.

vejpasop, 2021-07-02:
@mpvader Yes,

@izak. The only new dbus path for this is  `Ac/State/SplitPhaseL2Passthru` (See issue description)

izak, 2021-09-06:
Closing now that the dbus path is documented.

closed 2021-09-06 by izak