victronenergy/venus #832: Frequent "No Grid Meter Alarm: Warning" - Inverter Passthru - CCGX

NicholasHerbst, 2021-06-23:
Several installations have illustrated performance issues where a Victron Color Control GX (CCGX) is deployed. The occurrence of "Passthru" events appears to have increased with the Venus OS updates. 

A few sites have been resolved immediately with the introduction of a Cerbo GX to replace the CCGX - but this is not a sustainable solution. 

Clients have become more aware of the Passthru events occurring because Venus OS updates have introduced dedicated warnings/notifications to alert the user when one of these events is observed.

mpvader, 2021-06-24:
Hi @NicholasHerbst this might be a Venus OS issue; or might not be. Could you please take it through “the proper channels”? Which is either to Community or through your dealer/Victron sales manager. Sorry to redirect you like that, but this issue tracker is used for development only.

Have a good day, Matthijs

closed 2021-06-24 by mpvader

NicholasHerbst, 2021-06-24:
Hi @mpvader - understood. I'll attempt to drive the discussion through those channels.