victronenergy/venus #833: Have problems installing the touch screen...

KamiSchami, 2021-06-24:
Hi, I'm not getting the touch screen installed.  I'm following the instructions, but only bring error messages.  Please help me.

kwindrem, 2021-06-24:
What error messages are you getting?

What Raspberry PI model are you installing on?

Did you install SetupHelper also? It's needed for the RpiDisplay script to run.

KamiSchami, 2021-06-24:
I hava a 3B. What Setuphelper? I have manual here...

First i habe to install opkd Update, but then comes this error

[image]

[image]

KamiSchami, 2021-06-24:
[image]

All is working, but mit the touch

kwindrem, 2021-06-24:
Disregard my question about SetupHelper. I had assumed you were using my RpiDisplay package.

It isn't clear where you are getting the instructions to follow. I seem to recall some issues with opkg recently but don't recall the details.

I can offer an alternate way to make the Official Raspberry PI 7" touchscreen and potentially others work that doesn't require installing all the things I've seen in other posts on this community:

https://github.com/kwindrem/RpiDisplaySetup

This package does use SetupHelper and in fact, installing that package simplifies installing RpiDisplaySetup.

mpvader, 2021-06-24:
Hi all, sorry, but could you please continue this here:  https://community.victronenergy.com/spaces/31/index.html.

we use this issue tracker for development only.

closed 2021-06-24 by mpvader