victronenergy/venus #836: [Security] Root login by default

deosrc, 2021-07-13:
I am using VenusOS on a Raspberry Pi for testing purposes and it seems that the default account setup is against security best practices:

- Root login should be disabled with all access which requires root done through `sudo`
- A standard user should be configured for normal access (e.g. `pi`)
- A default password may be provided but it should be forcibly changed on first login
- Auto-login should be disabled by default
- The documentation should recommend changing the password

As it is at the moment, there are likely many devices running VenusOS with unsecured root privileges available by default.

It would also be beneficial to add a security policy to this repo so that vulnerabilities such as this can be highlighted in private while they are addressed: https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository

jhofstee, 2021-07-16:
- ssh and root login is disabled by default, unless you have physical access
- there is no password at all, you need to set it first or have physical access
- autologin is fine for physical access, you can take out the sdcard as well to get access
- you have to change the password, because the root user doesn't even allow logging in with a password by default

closed 2021-07-16 by jhofstee

deosrc, 2021-07-16:
Sorry but I do not agree with the above. The only barrier to root level access should not be physical access. Even in data centres with strict physical access controls, root auto-login would be a major security issue. VenusOS may be used in remote locations where physical security is not easily monitored or controlled. Breaches of security (and possibly damage to hardware due to malicious configuration) could be a very real risk if the OS is not secured correctly.

As a minimum, the documentation should be specific with a warning that root without any credentials is configured by default for physical access and that steps should be taken to secure this if physical security is insufficient.

The point of adding a security policy to the repo has also not been addressed.

jhofstee, 2021-07-17:
> Sorry but I do not agree with the above.

Fine with me, but it just shows you have no idea what you are doing.