victronenergy/venus #838: html5-app: add venus version and build date

martinbosma, 2021-08-11:
For the HTML5 app we would like to show the VenusOS version and build date.

Options:
- Add DBus paths; this makes it available for other services and as well as mqtt and modbus-tcp
- Make the version file available to the apache document root, so the html5 app can read it

jhofstee, 2021-08-31:
The version file is already available as /venus/version on the webserver.

martinbosma, 2021-09-01:
But we're still keeping the DBus paths from https://github.com/victronenergy/dbus-systemcalc-py/commit/b694bdeda60555d766c5b8807c6907f4d028ed4c?
The HTML5 app uses the equivalent MQTT topic atm. Having this info on DBus makes it available for other services, too.

closed 2021-09-02 by jhofstee