victronenergy/venus #842: Add Home Assistant integration

mpvader, 2021-08-17:
Izak has Home Assistant in his home; and Home Assistant people are doing a push for energy management.

Louisvdw, 2021-08-18:
Is the idea that this will become part of the Large Image version?

izak, 2021-08-18:
@Louisvdw No, this is about adding a component to [HA][1].

[1]: https://github.com/home-assistant/core

Louisvdw, 2021-08-18:
OK so this will be a seperate HA installation. Makes sense as I think HA can be cpu intensive.

JladanZA, 2021-09-18:
looking forward to this integration with eagerness

apple047, 2021-09-18:
I'm looking forward too, been trying to map modbus address in HA and it worked well for reading values but I can't write anything like the grid setpoint, no luck via MQTT either, I can only get the serial number of the GX device.
Having an integration will be a great benefit for many people.
Thanks in advance

JladanZA, 2021-09-18:
yup i got the same setup, all my sensor are working and have set min soc setup, but would be great to extend it

mpvader, 2021-10-21:
Hi! I recommend either MQTT, and if thats inconvenient for HA, then ModbusTCP.

for mqtt, see our dbus-mqtt repo

aaronsb, 2022-08-27:
MQTT + Venus is easy and adds lots of sensors. I think if anything, a superficial integration that simply packages the MQTT keepalive and a set of common sensors would be a good start instead. This is what I have in my HASS setup but never bothered to make it an installable integration. HACS might be a good place to start with this.

jon-nfc, 2022-11-03:
here's my 2 cents on how I integrated venus with HA.

Venus OS is installed on a raspberry pi. I'm using the large image. the venus PC and HA are on a separate iot wifi network.

initially I turned on venus mqtt service, however it's in the wrong format for HA. so i turned it off.

I enabled node red, and built a flow that reads the sensor data from venus (uses victron nodes), builds the mqtt discovery and state packets and sends it via mqtt to home assistant. with this setup, I have my BMV, MPPT controller and the venus system showing up as devices in home assistant. from there you can access the data the same as any other HA entity.

I've even built a node red flow that adjusts DVCC so that my batteries (Lead Acid) only ever receive a current of 0.1C. this way whenever I add load, DVCC adjusts MPPT to output more current.

> Interested in helping with this. My first thought was to get the Victron HTML5 screen showing as a dashboard in HA, but I'm sure there are ton of other details that can be extracted an graphed. What is the best way to interact with venus os? SNMP? Is there an API?

I have the venus touch display (the html panel) in HA as it's own panel using a html webpage card to http://venus.local.

drtinaz, 2025-11-28:
There is also a very good victron HA integration here https://github.com/tomer-w/ha-victron-mqtt