victronenergy/venus #843: systemcalc: get AC volts, amps and frequency to the UIs

mpvader, 2021-08-17:
There are multiple solutions.

The reason for this is the VRM Dashboard, which is currently being extended by Teun to have far more data in its detailed-mode.

Discussed on slack here: https://victrondevelopment.slack.com/archives/C7C1ZU6UD/p1629140050083000

izak, 2021-08-18:
# Background:

* Systemcalc already knows which meters it will use for grid and genset: https://github.com/victronenergy/dbus-systemcalc-py/blob/master/dbus_systemcalc.py#L703

* If no grid meter and/or no genset meter, it falls back to vebus

* When using vebus for grid and generator values, it can only give you information about the one that is currently connected, so in this case you must unavoidably lose information about the other input.

# Specification

* Create a set of dbus paths under `/Ac/Sources/Grid/...` and `/Ac/Sources/Genset/...`, with some suitable representation below it.
* VRM can't really do anything with a service name, it will need the service type and the device instance in order to construct an MQTT path. Other local dbus services will find the service name itself more useful. So I suggest:
  * /Ac/Sources/Grid/ServiceName
  * /Ac/Sources/Grid/ServiceType
  * /Ac/Sources/Grid/DeviceInstance

* And the same for Genset.

# How it will work:

Below, "grid meter" means an energy meter used to measure the grid, and "genset meter" means an energy meter used to measure a genset. Subtree refers to the part under /Ac/Sources that provides information about that source.

1. If you have both a grid meter and a genset meter, then the Grid subtree will contain the information for the grid meter, and Genset will contain  the information for the genset meter. Easy peasy.
2. If you have meters for neither grid nor genset, then only one of the subtrees will be populated depending on the active input of the Multi/Quattro. The other one will be invalid.
3. If you have a grid meter, but no genset meter, then the Grid subtree always points to the grid meter, and the Genset subtree points to vebus ONLY when the active AC input is configured as a generator, otherwise the Genset subtree is invalid.
4. Conversely, if you have a genset meter, but no grid meter, then the Grid subtree points to vebus ONLY when the active AC input is grid, otherwise it is invalid. The Genset subtree always points to the genset meter.

AlexImb, 2021-09-13:
@izak can we add data attributes for these new paths? Most of the logic in VRM relies on data attributes that are then mapped to MQTT paths.

izak, 2021-09-14:
@AlexImb I added the attributes to vrmlogger, and I made an issue documenting them as usual in the VRM tracker.

mpvader, 2021-12-15:
This is all done and works.

closed 2021-12-15 by mpvader