victronenergy/venus #844: Lost Battery History Widget in VRM after loading 2.72 on Pi

jwarchol, 2021-08-17:
I recently made a new SD card with the 2.72 image and fired it up. Everything proceeded as expected, except several VRM widgets no longer show up, notably the Battery History widget (number of cycles, deepest discharge, etc). If I go in the remote console the values are all still in my SmartShunt, but they don't seem available in VRM. Has anyone else seen an issue like this before? Suggestions for what I'm doing wrong are gratefully accepted.

mpvader, 2021-08-18:
Hi @jwarchol its possibly a bug in VRM. I’ll find out and let you know.

jwarchol, 2021-08-18:
Here are some screenshots to show  what's happening there: 
[image]

This is VRM installation 111929

mpvader, 2021-08-19:
Hi @jwarchol , the Battery history widget is only available if "H7 - Minimum battery voltage" is available in the data. And its missing on your system; please restart your GX device, that would solve it.

jwarchol, 2021-08-19:
Restarting worked, thank you @mpvader 🎉 

Hopefully this Issue showing up in search helps folks with the same problem in the future.

closed 2021-08-19 by jwarchol

mpvader, 2021-08-19:
yes indeed. For future things, please though use Community: https://community.victronenergy.com. And then either the normal area or the Modifications area.

Making issues here triggers emails to all or almost all our developers, and so forth.

thanks!

jwarchol, 2021-08-19:
I did post there but it was quickly buried with the volume of new posts. Since the only change on my side was the release of the pi image, I opted to raise the issue. Apologies if too many people were notified. I know there are challenges managing notifications on a public repository like this. Appreciate the help!