victronenergy/venus #845: ESS "Charge Schedule" uses UTC time, not local time

magpern, 2021-08-18:
I set the ESS Charge schedule to charge at 02:00 for 3 hours. It started 04:00 local time
I am 2 hours ahead of UTC

It was found in "VenusOS 2.80-11-large-20", but could probably be a bug in other versions too.

[image]

[image]
(x-axis is local time)

mpvader, 2021-08-18:
Hi magpern, thank your for the detailed report! This issue was introduced earlier in v2.80 development, and has been fixed as well now. There is a new version available, v2.80~13:

https://community.victronenergy.com/questions/98618/venus-os-v28013-available-for-testing.html

For future reports, can I ask you to add a reply to community, rather than make an issue? Then we have it all in one place.

closed 2021-08-18 by mpvader

magpern, 2021-08-18:
I need the large-OS, so I'll wait for it, and not use Charge Schedule in the meantime