victronenergy/venus #851: dbus-vecan/gui: Lynx Smart BMS finishing up

mpvader, 2021-08-25:
One thing that is open is to use the newly defined vregs.

currently publicly available Lynx Smart BMS firmware, v1.03, already contains those new VREGs. It has both the old and the new type.

VictronConnect already uses the new ones.

Fixes that most of the LSB the alarm & warning messages are confusing to the user.

Note: for Venus OS, do not implement W-B06, load will disconnect. And also not A-B06, Load disconnected. Doing so will cause double entries into alarm logs and such.

Roland has a document with all error codes & descriptive texts to show to the user; which is the same as now being added into the manual as well as in VictronConnect.

martinbosma, 2021-11-30:
Planning to introduce the following new battery alarms:
- `/Alarms/HighCurrent` - Similar to `/Alarms/HighDischargeCurrent`, `/Alarms/HighChargeCurrent`, but without direction information; alarm + warning
- `/Alarms/Contactor` - Bad contactor; warning only
- `/Alarms/BmsCable` - BMS cable fault; alarm only
- `/Alarms/HighInternalTemperature` - High BMS temperature; also used by the Lynx Shunt VE.Can; warning only

See [here](https://github.com/victronenergy/venus/wiki/dbus#battery) for the dbus defintion for a battery.

martinbosma, 2021-12-01:
Since VenusOS was lacking proper alarm support for the Lynx Shunt BMS, it issued error codes to inform the user that something was wrong. These error codes have never been registered in velib until [now](https://github.com/victronenergy/velib/commit/568eee0e6ae3885da1666fd89564a13c681d1737). To avoid double entries in the alarms logs, we simply suppress these error codes in the ranges `101-113` and `201-224`

martinbosma, 2021-12-06:
@mpvader & @rjacobsVE 
My proposal for the distributor & fuses paths:
```
/Distributor/A/Status                <= 0=Not available, 1=Connected, 2=No bus power, 3=Communications Lost
/Distributor/A/Alarms/ConnectionLost <= 0=Ok, 2=Alarm
/Distributor/A/Fuse/0/Name           <= UTF-8 string, limited to 16 bytes in firmware
/Distributor/A/Fuse/0/Status         <= 0=Not available, 1=Ok, 2=Blown
/Distributor/A/Fuse/0/Alarms/Blown   <= 0=Ok, 2=Alarm
/Distributor/A/Fuse/n/*              <= Repeat for every fuse in the distributor
/Distributor/B/...                   <= Repeat for every distributor (A-E)
...
/Distributor/E/...
```

This requires new VREGs for the 2-bit fuse status per distributor and possbily also for the number of fuses a distributor supports. Current hardware can hold up to 4 fuses, but new hardware might support more.

mpvader, 2021-12-06:
Hi @martinbosma, lets make the status "Not in use", rather than "Not available". Not in use is more clear to the user. Not available is (too) ambiguous,. Otherwise, no comments. Nice.

martinbosma, 2021-12-06:
I've updated the fuse status bits to 4 options, so we can distinguish between fuse slot not available (`0`) and not used (`1`). If a distributor supports only 4 fuses, the status of fuse 1-4 will read > 0 and 5 onwards will read 0. This makes a register with the number of fuses for a distributor obsolete.

mpvader, 2021-12-15:
moved to v2.90 milestone

martinbosma, 2022-01-03:
Some screenshots:

Battery page shows the Fuses submenu if `/NrOfDistributors > 0`
[image]

Top level Fuses page; showing an overview of all distributors with a summary status
[image]
[image]
[image]
[image]

Distributor page; showing the status of the fuses (if the distributor status is ok)
[image]

Distributor page; fuses are not shown if the distributor status is `No bus power` or `Communication lost`
[image]

martinbosma, 2022-01-03:
Fuse blown alarms; showing the name of the fuse in the notification:
[image]

Distributor connection lost alarm:
[image]

martinbosma, 2022-01-07:
@izak I've created two new tasks to add distributors and fuses support to VRM. Can you take care of that?

martinbosma, 2022-01-07:
@mpvader With the new changes to support the alarm and distributors, new LSB firmware (v1.04) is needed to be able to see the distributor and fuse status in the GUI (for firmware < v1.04, the fuses menu is hidden). 

The [newly added alarm paths](https://github.com/victronenergy/venus/issues/851#issuecomment-982730260) will work for firmware v1.02 an newer, so no issue there.
However, since the change also [suppresses the errors codes](https://github.com/victronenergy/venus/issues/851#issuecomment-983538392), the user will no longer receive fuse blown error notifications when they update VenusOS to the version that includes this change. 

We could:
- Advice to update to firmware v1.04; and make sure that this is available before we release VenusOS with distributor support
- Conditionally suppress the error codes, based on product id and firmware version (rather not)

Please advice

mpvader, 2022-01-07:
Option 1 is fine. And there is no need to advice that in the gui, but if you can easily do that: welcome to.

martinbosma, 2022-01-07:
Ok, I will not take further action then

martinbosma, 2022-08-19:
>verify that remotely controlling the relay works. And if it does, add that to both dbus wiki page as well as Venus OS large manual, section 'relays'. For both of those pages its fine to add that while v2.9 is not officially released yet.

The LSB does not support manual relay control at the moment. Roland added it as feature request.

closed 2022-08-19 by martinbosma

mpvader, 2022-08-21:
ok good. I've added it to the public docs here:

https://www.victronenergy.com/live/venus-os:large#controlling_relays

nmbath, 2022-08-21:
Could this [link](https://github.com/victronenergy/venus/wiki/dbus#battery) be updated, as there is no reference to the valid values for Mode or Status.