victronenergy/venus #854: NMEA 2000 Vendor name

DevoDave, 2021-08-28:
NMEA2000 labeling of the Venus product and devices is not accurate.

Vendor comes up as "1.0" for my CerboGX and the devices come up as unknown-n2K-device (or similar.. working from memory.)

Should hopefully be simple to fix and just require some labels added?

mpvader, 2021-09-30:
Hi @DevoDave , thanks for reporting. Please tell me what software you use in which you see this information?

DevoDave, 2021-09-30:
Raymarine Lighthouse 3 most current version.

virtuvas, 2021-09-30:
odd, on my setup with around 20 N2K devices all are fine except for the Model which comes up as 1.0. Both  manufacturer and name are spot on!

That is a raspberry pi 3B+, a BMV700 and a Bluesolar MPPT 30/100
I'm running the large built with SignalK (excuse the salt deposits on the Garmin, came back 3days ago and haven't cleaned up yet)

[image]

[image]

[image]

[image]

[image]

mpvader, 2021-12-15:
hi both, this is just a note that I've seen it. Sorry for not responding earlier, and thanks for reporting!

I have to warn, it might take a while before we look into this.

martinbosma, 2022-05-20:
Have checked this with Simrad, Garmin and Raymarine MFDs and also with Actisense NMEAReader and Maretron N2KAnalyzer and it looks like everything is according to NMEA2000 specification. I did see some "unknown-n2K-device" lables on the Raymarine, but this appears to be a (display) issue on their side, not on our side.

I did notice that most MFD's show the "Manufacturer Model version" field as "Model", while it actually is a hardware version, hence the (fixed) 1.0. The "Manufacturer Model ID" contains the model name, e.g. "BMV-700". All tested MFD's displayed this correctly.

All seems to be working as expected, so closing this issue.

[image]
[image]
[image]
[image]

closed 2022-05-20 by martinbosma