victronenergy/venus #857: Increase range of resistances for tank sensors

AlexMuir, 2021-08-31:
Currently the maximum resistance that can be saved for tank sensors is 264Ω.  The Cerbo has no problem displaying the actual resistance value in the setup menu (up to 330Ω) so the hardware has the range.  

The resistance range for standard Vetus senders is  0Ω (full) to 300Ω (empty).  If the resistance range can be increased, then Vetus sensors will be fully supported out of the box.  

Reference: https://www.vetus.com/en/waste-water-sensor.html

mpvader, 2021-09-24:
Hi @AlexMuir , thanks for this. We can increase it to 300Ohm, and will do that. all the best, Matthijs

AlexMuir, 2021-09-26:
Wonderful, thanks!

closed 2021-10-05 by jhofstee

mpvader, 2021-10-15:
Hi, available per v2.80~24, in candidate since today. thanks again.