victronenergy/venus #858: Add support for Ruuvi wireless temperature sensors

mpvader, 2021-09-06:
as discussed, some inspiration is in [mpvader/dbus-ruuvi](https://github.com/mpvader/dbus-ruuvi/blob/main/dbus-ruuvi), which uses code from [tty/ruuvitag-sensor](https://github.com/ttu/ruuvitag-sensor/tree/master/ruuvitag_sensor).

Features:
- giving sensors a name (custom names)
- temperature types (just like we already have)
- ignoring/selecting sensors  (the sensor from the neighbour shouldn't suddenly be added)
- read-out temperature, humidity, pressure, battery status; and anything else relevant (if there is more that is relevant).
- make sure to publish some identifier that matches with what the user can see in the ruuvi app or on the ruuvi device
- no need for alarms; as we don't have alarms on the Cerbo-built-in temperatures yet either.

Data example:
```
# MAC C3:E9:A1:E8:C9:C1
# {'battery': 3157.0, 'pressure': 1019.17, 'mac': 'c3e9a1e8c9c1', 'measurement_sequence_number': 18899, 'acceleration_z': 976, 'acceleration': 979.61625139643331, 'temperature': 17.550000000000001, 'acceleration_y': 76, 'acceleration_x': -36, 'data_format': 5, 'humidity': 37.210000000000001, 'tx_power': 4, 'movement_counter': 62}
```

mpvader, 2021-09-22:
Here is the information shown from one sensor. Note that if it has atmospheric pressure, then that's visible too:

[image]

The Setup submenu has one option, allowing to select the type:

[image]

The Device submenu shows various info:

[image]

Settings -> I/O menu has a new submenu:

[image]

In that submenu you need to select which sensors to connect to:

[image]

mansr, 2021-09-27:
Included in v2.80~21.