victronenergy/venus #859: Inverter RS: Sync SOC and Extra Charge Current

izak, 2021-09-10:
For the Multi this was done a long time ago. The same is needed for Inverter RS:

- [ ] If there is a BMV or another managed battery in the system with a more accurate SOC, copy this to /Soc on the inverter
- [ ] If there are other solar chargers in the system, write their combined charge current to /ExtraBatteryCurrent on the inverter

On VE.Can, the inverter already knows about the charge current of other VE.Can connected devices, and we need only inform it of what the VE.Direct chargers are doing (and maybe ve.bus).

rvanzundert, 2021-09-10:
The battery state-of-charge can be communicated to the Inverter RS and Multi RS using VE_REG_SOC (0x0FFF), un16 = SOC [0.01%], 0xFFFF = not available.

On ve.can the the RS units listen for VE_REG_SOC broadcasts from:
- lowest priority: a battery (e.g. Lynx Smart BMS)
- highest priority: a GX device
Alternatively the VE_REG_SOC can be written directly to the unit, in this case it will not listen for broadcasts anymore. This can be used on ve.direct or a ve.smart network if needed.

Receiving a state-of-charge in this way overrules the internal state-of-charge mechanism.
The timeout is 1 minute, so after 1 minute absence the internal mechanism takes over.

martinbosma, 2021-12-08:
dbus-vecan will support two new paths on `com.victronenergy.vecan.canx`:
- `/Link/Soc` - periodic broadcast of `VE_REG_SOC` with 60s time-out
- `/Link/ExtraBatteryCurrent` - periodic broadcast of `VE_REG_LINK_EXTRA_BATTERY_CURRENT` with 60s time-out

mpvader, 2021-12-15:
@izak pls confirm (test!) whether this all works now, both for Inverter RS and Multi RS, or if it still needs work.

izak, 2021-12-16:
No this definitely still needs work!

mpvader, 2021-12-16:
prioritise this, should be simple - or not?

closed 2021-12-17 by jhofstee