victronenergy/venus #861: v2.80: Possible slowness related to MQTT

mpvader, 2021-09-21:
There is a thread about this on Community.

First step is triaging, which Izak will do.

https://community.victronenergy.com/questions/100658/venus-os-v28016-available-for-testing.html?childToView=102531#comment-102531

izak, 2021-09-23:
OK, this one is confirmed. It seems that if you send a large enough payload with the keepalive (about a hundred values or so is enough on a slow platform), and if you send it often enough, it ends up spending all its time dealing with parsing the keepalive and very little (or no) time publishing actual values.

mpvader, 2021-09-27:
Next for this is comparing v2.73 vs latest v2.80 wrt cpu idle and dbus-mqtt cpu load in particular. Which Izak will do, whereafter we can see if more work on this is necessary.

More details in discussion in Slack, also on how VRM is currently using the keepalives.

mpvader, 2021-12-15:
Fixed, in two steps: first we improved how dbus-mqtt handled large keepalives.

And thereafter we fixed VRM to not use the keepalive system anymore, since its nonsense for it to use that.

closed 2021-12-15 by mpvader