victronenergy/venus #865: Data from MultiPlus Compact remains on overwhile while off

mpvader, 2021-10-01:
https://community.victronenergy.com/questions/103648/bug-ac-output-information-still-shown-with-inverte.html

Note that its a MultiPlus compact, something with a dipswitch (?)

And this shots were taken while in the situation. The data is made available on the D-Bus by vebus-interface:

[image]

[image]

[image]

[image]

[image]

kwindrem, 2021-10-01:
FYI:
uninstalled GuiMods - no change
problem exists in v2.80~21, v2.73, v2.65
upgraded to VE.Bus 487 (from 485) - no change

kwindrem, 2021-10-01:
More info:
When the AC out values change to --:
  the VE.BusAC  output dbus parameters change to -
  the VebusService in system changes to -, as to any related parameters (in system)
  system ACConsumption power and NumberOfPhases change to -

When the AC out values go to 0,
  the VE.BusAC  output dbus remain valid
  VebusService remains pointing to the MultiCompact service
  system ACConsumption power and NumberOfPhases remains valid

vejpasop, 2023-05-01:
"Fixes" where reverted in mk2-dbus v3.55 since they caused constant state toggles when in charger only mode while AC input was disconnected.

mpvader, 2023-05-01:
Hi. To make sure what you mean with “fixes”: are all changes that close this issue were reverted ?

vejpasop, 2023-05-01:
Yes, All changes belonging to this issue where reverted. 
I need thought I found a simple fix for this issue. But this approach seems to be "kansloos". So I need to reconsider.

mpvader, 2023-05-05:
Ok. then lets forget about it for now - plenty of more urgent things.