victronenergy/venus #866: Venus OS Large: No relay or IO capabilities on rpi4

SergioRius, 2021-10-03:
The menus for driving relays and IO pins access are missing on rpi4.

Also, If one tries to do a workaround in node-red by installing the stock node-red-node-pi-gpio it automatically gets deactivated, as it seems that the SO misses the required components for node-red to identify the raspberry hw and to drive the pgio.

Message shown:
`rpi-gpio : Raspberry Pi specific node set inactive`

mpvader, 2021-10-05:
hi, yes that standard node-red-node-pi-gpio pallette won't work. I assume its designed to run on Raspbian or something; which is not Venus OS.

I saw some other messages elsewhere from you as well: I suggest you first make the IOs work normally (ie via the Venus OS gui), and thereafter check Venus OS large. For any discussion or support on that, please use the Modifications section on Community. Thanks!

closed 2021-10-05 by mpvader

CountParadox, 2022-02-14:
https://community.victronenergy.com/questions/120771/venus-os-large-enable-relays.html

FWIW Ive got this working, see post for details!