victronenergy/venus #871: Feature request: Change dashboard background color

fpugliano, 2021-10-30:
When sailing at night, the white background of the dashboard disturbs my night vision.  Would it be possible to add an option to cange the background color from white to black?

[image]

mpvader, 2021-10-30:
Hi, yes! We’re working on a complete new UI, having a dark mode will be part of that.

closed 2021-10-30 by mpvader

fpugliano, 2021-10-30:
Awesome! Is this going to be in the next release?  When it is scheduled to roll out?

mpvader, 2021-10-30:
Hi, that will take a while, its a complete redesign of the UI. Somewhere next year. In case you’re on the northern hemisphere, then - with a bit of luck!- right when the common season starts.