victronenergy/venus #874: TSense doesn't work with VE.Can solar chargers

izak, 2021-11-19:
When using STS to pass a temperature value from a temperature sensor on the GX device to a VE.Can solar charger, the solar charger will receive a zero value.

The behaviour is that if an integer is written to `com.victronenergy.vecan/Link/TemperatureSense`, it sets LINK_TSENSE (0x2003) to zero.

```
can0  1CEFFFE1   [8]  66 99 03 20 00 00 00 00
```

But if a float is written, it does the right thing (23.01°C in this case):

```
can0  1CEFFFE1   [8]  66 99 03 20 FD 08 00 00
```

TODO:
- [ ] systemcalc should write a float to /Link/TemperatureSense (ibu)
- [ ] dbus-vecan probably shouldn't send a zero if it gets the wrong data type (jho)

martinbosma, 2021-11-19:
@izak I can easily convert  `/Link/TemperatureSense` to float before sending and that seems to work:
```
# dbus -y com.victronenergy.vecan.can1 /Link/TemperatureSense SetValue %"float(25)"
  can1  1CEFFFE5   [8]  66 99 03 20 C4 09 00 00
# dbus -y com.victronenergy.vecan.can1 /Link/TemperatureSense SetValue %"int(20)"
  can1  1CEFFFE5   [8]  66 99 03 20 D0 07 00 00
```
Or should I ignore it if the type !- `float`? 

And how about `/Link/ChargeVoltage`, `/Link/VoltageSense` & `/Link/BatteryCurrent`? 
Should they be able to accept an integer value?

izak, 2021-11-19:
It is probably fine to ignore values that are not of the right type. It is better than to send zero.

It is not necessary to do the same for VoltageSense and friends, as long as an incorrect type (string, etc) at least doesn't cause some weird value to be sent.

The reason this specific item is affected is because dbus-adc sends an integer, and this is copied directly across.

The main part of the fix is the one I already did: I force it to be a float. I just think it would be safer (in addition) to ignore writes of the wrong type.

martinbosma, 2021-12-08:
There is still a [pending patch](https://github.com/victronenergy/velib/commit/c10f781efaef8a1a3a951406475c10a25f745275) in velib that addresses the issue when and empty or unparsable string is written to a dbus item.

If this is accepted, it will tick the last todo item.

mpvader, 2021-12-15:
@izak when was this bug introduced? In v2.80? Or before?

mpvader, 2021-12-15:
Also, its now fixed - right?

And we'll leave the issue open to not forget about that velib commit by Martin.

izak, 2021-12-16:
Yes, it is fixed. It was probably broken as far back as Venus 2.40, when the functionality for using an GX-connected sensor was introduced. From the beginning in other words.

It only affects the specific combo using a temperature sensor connected to the GX device, STS turned on, and VE.Can connected solar chargers.

izak, 2021-12-20:
My assessment of when this was broken was incorrect.

The breaking change in velib is around August 2019, which was not used in Venus 2.40 or Venus 2.50. The first official release with this problem was Venus 2.60.

martinbosma, 2022-08-22:
> There is still a [pending patch](https://github.com/victronenergy/velib/commit/c10f781efaef8a1a3a951406475c10a25f745275) in velib that addresses the issue when and empty or unparsable string is written to a dbus item.
> 
> If this is accepted, it will tick the last todo item.

@jhofstee There is still a pending patch, do you want to include this or not? Pls update the state of this issue accordingly.

closed 2023-07-28 by izak