victronenergy/venus #877: new option "Passthru Only"

nipperCZ, 2021-11-25:
Hi,
is it possible to add the option as "passthru only" to the settings for multiplus II?

[image]

mpvader, 2021-12-15:
hi nipper, could you explain a bit when you'd like to use that mode? Ie for what use case?

nipperCZ, 2021-12-16:
Hi, maybe because I know what the weather will be like that day and I want to lighten the battery?

Odesláno z iPhonu

kwindrem, 2021-12-16:
Is Passthru Only really what is wanted? If the grid fails, you loose your AC output.

What I see as a better mode is "Charger Off". That is, if AC input is present pass it thru to AC out. If grid fails run the inverter to power the loads. That way, you aren't pulling power from the grid to charge the batteries when you know they'll get charged by solar.

nipperCZ, 2021-12-16:
I have the charger turned off. I don't use it at all. That's why I need 
what I described. Otherwise I have to go switch the grid manually in the 
switchboard.

VK

Dne 16.12.2021 v 9:36 kwindrem napsal(a):

nipperCZ, 2021-12-16:
For example, today. The batteries were at 20% capacity and I added a new 
one. The drive was running in pass-through mode before adding a new 
battery. I had to logically turn it off before adding it, and after 
reconnecting, it started in inverter mode. Batteries with lower capacity 
would run below 20%.

Dne 16.12.2021 v 9:36 kwindrem napsal(a):
> pass it thru

pwf-boat, 2022-07-21:
A Passthrough only option would be useful for me in the following scenario. In winter my boat is drained down unused. It has some small electric heaters running of shorepower. The only DC use is the monitoring kit, Cerbo GX and the router. I have enough solar to keep up with the DC demand so do not want the charger running. If shorepower fails I do not want the inverter to power the heaters. I do not have any sockets direct from the shorepower. Passthrough would solve this. Currently I set to charger only and accept I am paying for mains electric to run the charger and some of the DC load as disabling charging you then need to go into the inverter settings.

neoneddy, 2023-01-03:
Here is a use case for this in the RV industry in the summer.   When my RV is plugged into our 20 amp outlet and the air conditioner is using nearly 100% capacity of the input current limit.   Usually the Multiplus will switch to hybrid assist mode  when it's that close.  During the night this slowly depletes the battery.   If I set the AC input current higher, it will then start charging batteries as well as running AC loads and pop the breaker eventually.  The only way around this so far is to adjust the DVCC charge current to 0 .  It would be great to have this pass through option to allow solar to charge during the day .  If I set DVCC charge current to 0 I don't get that.