victronenergy/venus #880: Battery measurements: Multi name missing when checking after plugin and before reboot

mpvader, 2021-12-15:
Hi Jos, can you triage this? Ie try and reproduce? Could also be fixed meanwhile - I'm not sure.

https://community.victronenergy.com/questions/108453/venus-os-v28029-available-for-testing.html

[image]

vejpasop, 2021-12-16:
Hi @mpvader, 

The issue was most probably already fixed in Venus V2.80~30.

From the Venus change log
```
dbus-systemcalc-py v2.88 & v2.89 & v2.90
    Ensure that a disconnected VE.Bus service is correctly redetected as VebusService if it returns
```

I tested it on Venus V2.80~32 and it worked ok. Tested with an onboard MK3 as wel as a MK3-USB

mpvader, 2021-12-16:
Great. closing.

closed 2021-12-16 by mpvader