victronenergy/venus #881: Ruuvis adding /Status path and issue when removing / missing a sensor

mpvader, 2021-12-15:
Two things:

1. We need to add a /Status path. Its used by VRM to validate that there actually is temperature sensor input. It can't rely on the temperature for that, since that might be missing due to a sensor error. See [here](https://github.com/victronenergy/venus/wiki/dbus#temperatures) for what can be put on that path. Depending on how its implemented in the dbus-ble-sensors, one solution could be to always set 0 (OK). And when not OK then remove the service from the D-Bus - as that is how I expect its implemented now?  Some other variant, which includes using the 1 (Disconnected) state is also fine - then enabled sensors can remain visible on the Device List in the GUI as well, even though the sensor is gone. Which does have its advantages to a user.

2. (seen on v2.80~29 only!) There is an issue visible in the gui when removing a missing a sensor. It has been reported by Stefanie, second item in this screenshot:

[image]

from: https://community.victronenergy.com/questions/108453/venus-os-v28029-available-for-testing.html.

Then I tried to reproduce, I have one sensor that is sometimes connected and sometimes not - range issue most likely. And it happened to now be not connected. So first I clicked the button in the gui to get it removed. Then I went to Settings -> I/O -> Bluetooth, there disabled that sensor, and then things got a little messy like in this screenshot:

[image]

Going out and back in to the menu doesn't matter, list remains the same. Note that that is on v2.80~29, both on my side, and for 99% also on Stefanies side.

This is what the D-Bus shows:

[image]

mpvader, 2021-12-15:
added "Ruuvi Tag show "not connected" in irregular intervals"

mansr, 2021-12-16:
The blank items in the settings menu for vanished sensors are caused by `VeQItemTableModel` or something related. The same thing happens if a USB tank sensor is unplugged.

mansr, 2021-12-16:
Sensors are removed from dbus (and thus show as "not connected") if they have not been seen for 60 seconds. If the signal is weak, it's possible this timeout is too low. Increasing it is trivial, though.

closed 2022-01-12 by mpvader