victronenergy/venus #882: v2.80 documentation updates

mpvader, 2021-12-15:
- update Victron & Pylontech that per v2.80, DVCC is force enabled. See https://github.com/victronenergy/venus/issues/784.

millijuna, 2022-04-02:
If it would help, I would also be willing to offer up my boat as a test platform for this feature. Once completed, I'll have a system consisting of a REC ABMS, Cerbo GX, and WS500. (As well as a multiplus II, MPPT solar controler, smart shunt as an energy meter etc...). I'm also planning on putting the REC and WS500 on the BMS.CAN port (configuring them for 500kbps to match the Cerbo). I do have a Yacht Devices YDNB-07 to bridge the N2K between the two ports.

My own development skills are a little rusty, but I am a systems integration Engineer by trade, so am pretty good at figuring these things out.

mpvader, 2022-07-14:
Ess manual is done as well and now translation is starting.

closed

closed 2022-07-14 by mpvader