victronenergy/venus #884: Support shelly plugs & temp sensors

mpvader, 2021-12-22:


mpvader, 2022-01-12:
Temp sensor: https://shelly.cloud/products/shelly-humidity-temperature-smart-home-automation-sensor/

4 channel AC switch din rail mountable: https://shelly.cloud/shelly-pro-smart-home-automation-solution/

Most common plug: https://shelly.cloud/products/shelly-plug-s-smart-home-automation-device/

luciangpl, 2022-01-12:
Hi, let's include these :
https://shelly.cloud/products/shelly-1pm-smart-home-automation-relay/  - It's a 16A relay plus single phase meter
https://shelly.cloud/products/shelly-3em-smart-home-automation-energy-meter/ - this is a 3ph 120A/phase meter plus a 10A single phase relay

MichelleWhy, 2022-01-14:
And maybe Shelly Flood - https://shelly.cloud/products/shelly-flood-smart-home-automation-sensor/ - Flood Sensor for the boaters.

aaronsb, 2022-01-14:
I have a lot of shelly devices woven into our "fabric" right now through abstracted mqtt. Is there any more information about where I might go to experiment with the proposed device trees?

mpvader, 2022-01-19:
Hi all, we’ll order and work on all five of the above listed Shelly devices. Thank you.

@aaronsb I think its best to await the implementation; I don’t know all the D-Bus paths yet.

jhofstee, 2026-09-06:
Shelly integration was added in v3.70.

closed 2026-09-06 by jhofstee