victronenergy/venus #885: dbus-vecan: Improve support for N2K GPS-es

martinbosma, 2022-01-11:
Reported on community:
https://community.victronenergy.com/questions/114224/cerbo-gx-nmea-2000-device-config-in-out.html?childToView=114971#comment-114971

Currently GPS devices must have device class 60 (Navigation) and function 145 (Own ship position GNSS) to be recognized as GPS.

jhofstee, 2022-03-22:
yes, that is per NMEA2000 specification. GPS messages need to be send from a dedicated function (there is no data instance e.g.)

millijuna, 2022-05-09:
I'm also facing this. The GPS on my boat is my Vesper XB8000 AIS transponder. The Garmin Instruments on my boat, and my ICOM M506 VHF radio all pick up the position from the Vesper, without issue. I don't see why the Cerbo GX should be any different. I know the data is getting there because it shows up without issue in SignalK.

closed 2026-10-08 by citolen