victronenergy/venus #888: gui for systems in "kiosk mode": automatically go back after going to the notifications screen

mpvader, 2022-01-12:
This improvement is for systems where the user can not use the touch screen - aka kiosk mode: there are systems out there where the user cannot use the touch screen. For example rental hybrid gensets, where the rental agency simply doesn't want the user to do anything on the GX - and to accomplish that they've mounted it behind a window. And they then, for example, install a Digital Multi Control to allow just the bare necessity of control by the operator of the system.

There is one issue for such systems: if something triggers the gui to show a notification (ie a low battery alarm), the gui goes to, and then stays on the notification screen - even after the alarm was cleared. Is there some simple rule we could make to automatically go back to the previous screen - and then still show some exclamation mark blinking to indicate that something happened and hasn't been acknowledged yet - once the count of active alarms has dropped back down to 0?

Doable in todays gui? or better done only in the design for the next generation gui?

mpvader, 2022-01-13:
Hi Kevin, thanks. We’ll consider this as well. Matthijs

SturmUDrang, 2022-04-14:
Hi! We would also need an addition to the kiosk mode where the touch would be completely disabled on GX Touch  so users can't play with language or other settings that are not disabled in user mode. The screen would be forced to stay on the page set on LAN or remote. 

BTW Is there any quick workaround to disable touch from OS side?

jhofstee, 2022-05-20:
For sure, never jump automatically in menus, it is just really annoying ...

mpvader, 2022-05-20:
Fyi This is being fixed in the new gui, the Qt6 one