victronenergy/venus #889: Ruuvi: add alarm for battery voltage

mpvader, 2022-01-12:
See here for discussion on levels:
https://github.com/ruuvi/com.ruuvi.station/issues/335

mpvader, 2022-01-19:
if temp is less than -20C, low battery threshold = 2V
else if temp is less than 0C, low battery threshold = 2,3V
else low battery threshold = 2,5V

jepefe, 2023-12-11:
I will include this alarm in venus platform service.

mpvader, 2024-01-29:
@jepefe , pls fix it and add it to the old gui as well; not make it part of venus-platform. You're here in the office with Dirk-Jan this week, should be easy to test.

jepefe, 2024-02-01:
dbus-ble-sensors has been changed to use` /Alarms/LowBattery` instead of `/Status` and alarm was added to the gui.

Both are added to the todo.

mpvader, 2024-02-01:
so per this change, we'll actually have that alarm show up in the GUI as a notification. Instead of only show somewhere hidden in a menu. Right, @jepefe  ?

Can you post a screenshot of how the alarm looks in the gui?

jepefe, 2024-02-01:
@mpvader Now it is only shown in the notifications center but not in the sensor page. If necessary I can make it show also in the sensor page.

[image]

mpvader, 2024-02-02:
hey @izak can you take care of the vrmlogger change? the updated dbus-ble-sensors repo you'll need to test against is ready, on the todo. (but not yet in a build)

closed 2024-02-05 by mpvader