victronenergy/venus #891: clean checkout; make raspberrypi2-venus-image; => linux-firmware-rpidistro does not have required commit 00daf85

cons0l3, 2022-01-20:
Dear all,

I am still a newb when it comes to compiling venusos. I want to add a kernel firmware module for a wifi usb stick "rtk8822bu" to the image. I cloned the repo, and did all the things explained in the docs necessary to compile an image. Learnt a lot ;-)  But now I am failing (or better the build is) which the error:
```
ERROR: linux-firmware-rpidistro-0.0+gitAUTOINC+00daf85ffa-r0 do_fetch: Fetcher failure: Unable to find revision 00daf85ffa373ecce7836df7543c6ebe4cf43639 in branch master even from upstream
ERROR: linux-firmware-rpidistro-0.0+gitAUTOINC+00daf85ffa-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'git://github.com/RPi-Distro/firmware-nonfree')
```

If I have a look in the git repo `RPI-Distro/firmware-nonfree` there really is no commit with the given hash as requested from `sources/meta-victronenergy/meta-bsp/recipes-kernel/linux-firmware-rpidistro/linux-firmware-rpidistro_git.bb`

Could someone with deeper knowledge please advice? 
a) switch to commit with hash XY
b) forget about the entire thing, as the license stuff is now fixed at the source? I am not sure what the reasong for the special treatment is.

Thanks so much. I will try myself of course.

Cheers

mpvader, 2022-01-20:
Hi, could you ask this at https://community.victronenergy.com, and then Modifications section?

also there I can’t promise that someone can help you unfortunately, but this issue tracker is reserved for functional issues, ie dev work. Not dev support.

ps. Try checking for that commit in a branch other than master.

closed 2022-01-20 by mpvader

cons0l3, 2022-01-21:
understood. thanks for your support @mpvader.

if someone else stumbles upon this issue, I found an issue at RPi-Distro where a lot of people are fighting with the deletion of the master branch and missing commits as well:
https://github.com/RPi-Distro/firmware-nonfree/issues/20

I will look into it. If I come up with a solution for venus, I will either document it here, or figure out a way to provide PR or add a patch.

cons0l3, 2022-01-21:
as the `meta-victronenergy/meta-bsp/recipes-kernel/linux-firmware-rpidistro/linux-firmware-rpidistro_git.bb` seems to be a copy of the yocto projects receipe [recipes-kernel/linux-firmware-rpidistro/linux-firmware-rpidistro_git.bb](https://git.yoctoproject.org/meta-raspberrypi/tree/recipes-kernel/linux-firmware-rpidistro/linux-firmware-rpidistro_git.bb?h=dunfell&id=934064a01903b2ba9a82be93b3f0efdb4543a0e8) I have "upgraded" the receipe to the newest dunfell yocto version. It compiles. I will do some testing if my rpi2 boots. I have not figured out how to properly test such a change against all versions of rpi yet.