victronenergy/venus #895: Vebus: Add notification when system reset is ommited after DVCC is disabled

vejpasop, 2022-01-28:
Once DVCC is enabled the VE.Bus system relies on active control by the GX device. When DVCC is disabled the VE.Bus system must be reset though a system reset. If not the VE.Bus system will time out after 5 minutes and shut down for safety reasons

When DVCC is disabled the GUI shows "Make sure to also reset the VE.Bus system after disabling DVCC" for 5 seconds.
If one forgets to reset the VE,Bus system, we want to give notification. This reminder will be triggered by mk2-dbus once writing to the battery operational limit dbus paths has timed out for about 4 minutes.

vejpasop, 2023-04-11:
This issue becomes obsolete once https://github.com/victronenergy/venus/issues/1032 is implemented. 
The Multi will issue an alarm after 5 minutes of no BOL frames

jhofstee, 2026-09-09:
#1032 is implemented, so according to above this can be closed.

closed 2026-09-09 by jhofstee